bluerobotics/BlueOS #1004: bug: file-browser page not working on some browsers

ES-Alexander, 2022-05-03:
### Bug description

Attempting to access the file browser on some browsers just gives an endlessly spinning logo. The underlying service seems to work fine, it's just our wrapper page that's breaking.

|OS|Browser|Works?|
|---|---|---|
MacOS 12.3 | Safari | ✅
MacOS 12.3 | Orion | ✅
MacOS 12.3 | Chrome | ❌
Windows 10 | Edge | ❌

### Steps to reproduce

1. Open an affected browser
1. Load BlueOS 1.0.x
2. Turn on pirate mode
3. Go to File Browser via homepage (or Tools dropdown in sidebar)
4. \<panik\>
5. Go to underlying page (e.g. http://192.168.2.2:7777 or via Available Services in sidebar Tools dropdown)
6. \<profit\>

### Primary pain point(s)

No access to file browser makes debugging and file downloads/modifications more challenging. While the service can still be used directly as a workaround, most users won't know that's an option, especially since it's not yet clearly documented.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2022-05-03:
This was fixed in https://github.com/bluerobotics/BlueOS-docker/releases/tag/1.1.0-beta.1, can you check ?

ES-Alexander, 2022-05-03:
Sorry, I should have checked that before posting the issue, but couldn't find any issues about it so assumed that it wasn't already a known problem.

Just tested `1.1.0-beta1` and the issue does indeed seem to be gone.

closed 2022-05-03 by ES-Alexander