bluerobotics/BlueOS #1011: bug: cameras inconsistently appearing in QGC

ES-Alexander, 2022-05-12:
### Bug description

Brought up in [this forum post](https://discuss.bluerobotics.com/t/disable-automatic-video-display-on-qgroundcontrol/12109):

> QGroundControl with BlueOS will automatically display the video streams ... [but] the video changing function is not very reliable and having 3 cameras on our setup, most of the time only 1 or 2 show up and its inconsistent with which camera shows and which don’t.

### Steps to reproduce

1. Start multiple streams in BlueOS
2. Check QGC for available ones
3. Repeat with inconsistent results [NEEDS CONFIRMATION]

### Primary pain point(s)

Makes it harder to effectively use multiple cameras with BlueOS, and encourages using external applications for streaming any/all video instead of QGC (which fragments the user experience).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2022-05-12:
@Williangalvani said (internally)
> It smells like MAVLink routing to me.
> I think a good start is to set up 3 streams and check if the heartbeat rate in QGC is 1Hz for all cameras

Makes sense to be a MAVLink issue if the cameras are appearing fine in BlueOS and can be streamed without issues when using direct UDP ports (needs confirmation, but seems likely).

Deepwater-Exploration, 2022-05-13:
Hello!

Yes, the cameras appear fine in BlueOS and when using software such as OBS to receive stream, all 3 cameras show up with high reliability. The problem only appear with QGroundControl and the moment Mavlink camera shows up.

Thanks!

Deepwater-Exploration, 2022-06-02:
Interesting. Was there an update because I reflashed it before and same issue. Also it may be a bit different because I am using 3 cameras

ES-Alexander, 2022-06-02:
Turns out the issue was not in fact solved - it just temporarily worked for them after they flashed, and is still somewhat intermittent.

shgavman, 2022-07-30:
Hey guys, did anyone get to the bottom of this? We're also having trouble with multiple cameras in BlueOS.

patrickelectric, 2022-08-02:
We did investigate this issue, and from what we discovered it appears to be a problem with mavlink-router.

To explain it a bit better, for each camera on the camera-manager, we create two threads, one for the heartbeat and the other to deal with incoming mavlink messages. 
Once mavlink-router receives two different heartbeats from the same UDP socket, each heartbeat with a different mavlink camera component ID, mavlink-router starts to send just messages for the latest heartbeat component ID received from the camera-manager. As a result, only a single camera works, and if there is a time difference between multiple cameras heartbeat, the others may work resulting in video change between cameras on QGC.

joaoantoniocardoso, 2022-08-02:
In addition to what Patrick said above, I want to point out two other related things:
- The ordination in the QGC combobox for camera selection is inconsistent;
- The name of the cameras that are shown in QGC is not what is chosen in mavlink-camera-manager.

patrickelectric, 2024-06-19:
@joaoantoniocardoso is this still a thing ?

joaoantoniocardoso, 2024-06-20:
> In addition to what Patrick said above, I want to point out two other related things:
> 
>     * The ordination in the QGC combobox for camera selection is inconsistent;
> 
>     * The name of the cameras that are shown in QGC is not what is chosen in mavlink-camera-manager.
> 
> 
> Both degrade the experience for multiple cameras.

These both observations are still true