bluerobotics/BlueOS #1016: Bridget: hot-plugging is not working

Williangalvani, 2022-05-20:
We had a report that upon disconnecting and reconnecting a serial port, the bridge isn't rebuilt

patrickelectric, 2022-05-20:
The latest version of bridges now has an option to disable auto disconnect and also detect serial disconnection. 
https://github.com/patrickelectric/bridges/commit/16a7b82658823c1b9f119c887be38ea0475eb75e
https://github.com/patrickelectric/bridges/commit/459cf353ad7a7b5ace8fe6f941e8601caff62c87

Will create a PR for blueos soon.

rafaellehmkuhl, 2022-07-21:
@patrickelectric was this solved with the recent PRs?

patrickelectric, 2022-07-21:
No, this makes possible for bridget to restart bridges if something goes wrong on the serial port.

patrickelectric, 2022-07-21:
bridges itself now just closes if a serial port is disconnected.