bluerobotics/BlueOS #1031: bug: Update waiting window hangs while waiting for core to boot

patrickelectric, 2022-06-03:
### Bug description

[image]

### Steps to reproduce

Do any update with version chooser

### Primary pain point(s)

The user needs to update the page manually 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2022-06-07:
hmmm... time image is missing

ES-Alexander, 2023-01-16:
Confirming that this is still an issue. Would be nice if we can squash this one even if it's just with an auto page refresh after some amount of time or something (although detecting that BlueOS is back would be a nicer solution).

rafaellehmkuhl, 2023-05-30:
This issue got worse. The loading component use to disappear after some (long) time. Now it just stays there indefinitely.