bluerobotics/BlueOS #1033: One of those are not like the others (version chooser name conflict)

rafaellehmkuhl, 2022-06-03:
[image]

1. I'm running latest master from blue's main dockerhub repo
2. At some point I apparently updated my BlueOS version to master but from my personal dockerhub repo
3. Version-chooser thinks I'm currently running both images, although their hash is different.

rafaellehmkuhl, 2022-06-03:
a classic case of [image]

Williangalvani, 2022-06-03:
the "running" maker is not using the fullname. it should be an easy fix

JoaoMario109, 2025-01-27:
Closed by #1508

closed 2025-01-27 by JoaoMario109