bluerobotics/BlueOS #1041: The user should be warned when bootstrap reverts to "Factory" version

Williangalvani, 2022-06-08:
You may not notice it if you don't look at version chooser, which is a large oversight on our part

dwright213, 2023-01-29:
Is there a config file somewhere that can be set to make it always choose a specific beta, or the latest version that's been downloaded?

ES-Alexander, 2023-02-14:
> Is there a config file somewhere that can be set to make it always choose a specific beta, or the latest version that's been downloaded?

The behaviour is already to start out trying to run the latest `core` image that was applied, so if you turn off your vehicle having just run `1.1.0-beta.15` then that's the image that _should_ be run when you next turn on the vehicle. The "factory" image is only fallen back to if the latest run image fails (which should be extremely rare, unless you're developing `core` and happen to create and run a broken image).

For some insight into the current process:
1. The `blueos-bootstrap` container is run on system startup (via `/etc/rc.local`)
1. [`bootstrap`](https://github.com/bluerobotics/BlueOS-docker/blob/master/bootstrap/bootstrap/bootstrap.py) reads its startup configuration from `startup.json`
    - The file is stored on the base file-system, at `/root/.config/blueos/bootstrap/startup.json`
        - The `/root/.config/blueos` directory is mounted into the `core` container at `/root/.config/`, so `startup.json` can be accessed from within the `core` container (e.g. using the File Browser or Terminal pages in the BlueOS web interface) at `/root/.config/bootstrap/startup.json`
        - The BlueOS Terminal can also drop down into the base operating system via the `red-pill` command, in which case the file is found at its actual location in the file-system, although it requires elevated permissions to access
1. `bootstrap` tries to run the `blueos-core` container from the configuration it read (this is the latest used image)
1. if all goes well then bootstrap just continues periodically checking that the container is still running as expected, so it can restart it if it crashes
1. if something goes wrong (e.g. with reading the file, or starting `core`) then bootstrap replaces the current (non-working) `startup.json` with its internally stored [`startup.json.default`](https://github.com/bluerobotics/BlueOS-docker/blob/master/bootstrap/startup.json.default#L3)
    - this specifies to use the "factory" image
    - as far as I'm aware, this file is only stored within the `bootstrap` docker image (not mounted to the base file system), in which case it's not persistently modifiable
        - if this is actually modifiable then it would be one approach to change the default image, although doing so could be dangerous because you could accidentally delete the image you set as the default via the BlueOS Version chooser and end up with no fall-back

---

If you want a way to set the fallback image, I expect the most feasible (albeit semantically problematic) approach currently is to tag that image as the new `factory`, e.g. via the BlueOS Terminal page:
```shell
root@blueos:~# red-pill                             # drop down to the operating system
...                                                 # intro text
pi@blueos:~ docker image list                       # see available images
pi@blueos:~ FACTORY=1.0.1                           # choose a blueos-core image to be the new factory
# [Optional] save a backup of the current factory image:
pi@blueos:~ docker image tag bluerobotics/blueos-core:factory bluerobotics/blueos-core:factory-backup
# tag your desired image as the new factory image
pi@blueos:~ docker image tag bluerobotics/blueos-core:$FACTORY bluerobotics/blueos-core:factory
```

As mentioned, this is semantically not great, because "factory" implies that's the default image from Blue Robotics, which can be meaningful for support purposes. 

@Williangalvani / @patrickelectric perhaps we should revise bootstrap to have two fallbacks, one that's user-set (which could be directly selected via Version Chooser in pirate mode), and if that fails then fall further back to the "factory" image? I know that adds complexity, which we'd rather avoid having in bootstrap, but it does seem valuable for a user to be able to specify a fallback, especially given how much the interface has changed from when we started (it's quite jarring to end up in the old interface and have to redetermine where things are).

PR #1946 (2023-08-08): core: frontend: create SelfHealthTest, add test for factory images
[image]

fix #1041