bluerobotics/BlueOS #1046: bug: BlueOS FIRMWARE_FETCH_FAIL

magicbycalvin, 2022-06-13:
### Bug description

Version: tags/1.0.1-0-gbc26236

I installed BlueOS on what I believe is the R3 BlueROV2 (I am picking up a project where others left off so I don't have too much useful information about the system). I do know for sure that I am using a Raspberry Pi 3.

Anyways, when trying to use the firmware update tool, I am unable to see any firmware options and I notice a notification with the following text
```
Autopilot Manager
FIRMWARE_FETCH_FAIL
Not a gzipped file (b'<h')
3 minutes ago (1 times)
```

### Steps to reproduce

1. Install latest version of BlueOS (version number above)
2. Connect to BlueOS via IP address 192.168.2.2
3. Select Firmware
4. Select CHOOSE FROM REPOSITORY
5. For "Vehicle" dropdown menu, select Sub <--- Failure occurs at this point

### Primary pain point(s)

Unable to use the convenient firmware update tool.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-06-14:
Hi! Thanks for reporting!
Could you get the service log from the this service? You should hopefully be able to find it in `var/logs/blueos/services/ardupilot-manager/` via the [File Browser](https://docs.bluerobotics.com/ardusub-zola/software/onboard/BlueOS-1.0/advanced-usage/#file-browser).

PS: Does your ROV has access to internet? It's required in the case of firmware fetch from the ArduPilot cloud repository.

patrickelectric, 2024-06-19:
This should be fixed now with latest versions of blueos, better in 1.3

closed 2024-06-19 by patrickelectric