bluerobotics/BlueOS #1051: Video streams cannot be disabled, just deleted

rafaellehmkuhl, 2022-06-15:
We should be able to.

dgudiel, 2022-10-25:
further, a stream can't be deleted unless the camera is plugged in. 

If I disconnect a camera and plug in a different camera, I cannot create a new stream to the same endpoint. I have to connect the previous camera, delete the stream, then create the new stream for the new camera.

joaoantoniocardoso, 2022-10-26:
Thanks for bringing it up @dgudiel.

From my investigation, this is a problem on both BlueOS and mavlink-camera-manager sides.

For BlueOS, it is not showing the configured streams if the camera is not available.

For mavlink-camera-manager, it is not deleting streams if the camera is not available (I raised [this](https://github.com/bluerobotics/mavlink-camera-manager/issues/141) issue there).

@dgudiel for now, you can go to the mavlink-camera-manager developer interface (blueos.local:6020) and delete the stream there, without reconnecting the camera again.