bluerobotics/BlueOS #1072: install: Add support for Ubuntu 64-bit

ES-Alexander, 2022-07-13:
Raised in [this forum post](https://discuss.bluerobotics.com/t/raspios-or-ubuntu-arm64/12236/4):

> In case anyone else is attempting to install on Ubuntu 20.04, here are a few solutions to problems you'll encounter.
> 
> 1. rfkill isn't installed, which is easy to remedy.
>    ```
>    sudo apt-get install rfkill
>    ```
> 2. The default user isn't named 'pi', which causes errors with the install script. Can be fixed by following this guide: [https://www.linuxuprising.com/2019/04/how-to-change-username-on-ubuntu-debian.html](https://www.linuxuprising.com/2019/04/how-to-change-username-on-ubuntu-debian.html)
> 3. A few file paths are different, which require you to clone the BlueOS repository locally and make necessary changes.
>    - The path in [line 16](https://github.com/bluerobotics/BlueOS-docker/blob/0b6976f892bba2dfe5723b6a0373312c96dad22d/install/boards/bcm_27xx.sh#L16) of bcm_27xx.sh should be changed to:
>       ```
>       /boot/firmware/overlays/$DTS_NAME.dtbo
>       ```
>    - And [line 67](https://github.com/R-Stokke/BlueOS-docker/blob/1589dde77701fa702b6588a6a6ea9fd6ab27ea76/install/boards/bcm_27xx.sh#L67) to:
>       ```
>       /boot/firmware/cmdline.txt
>       ```
>    - And finally the paths in [lines 161 and 163](https://github.com/bluerobotics/BlueOS-docker/blob/886f486753ac68fbe092fb185ba505f3fcf5cba6/install/install.sh#L161) in the install.sh script should *both* be changed to:
>       ```
>       /etc/dhcp/dhcpd.conf
>       ```
> 
> I ended up opting to use 64-bit RaspiOS for my application, so **there may be more issues than are listed here**, but this might at least shave an hour or two off of your troubleshooting.

patrickelectric, 2024-03-11:
https://github.com/bluerobotics/BlueOS?tab=readme-ov-file#deveompment-environment helps a lot with that.
If the users adds the necessary binds and enable wifi, cable guy and commander, it may work.

NoahMollerstuen, 2024-10-12:
I've tried installing BlueOS on a Raspberry Pi 4B running Ubuntu 24.04. The docker container doesn't start on boot, but after starting it manually it sets up the network configuration correctly and I can access the web dashboard. The frontend shows a series of 502 error responses in the notifications tray. Do you have any idea what changes would be required to fully support Ubuntu @patrickelectric ?

and-human, 2024-12-07:
@NoahMollerstuen did you find a solution?

patrickelectric, 2024-12-07:
Hi, the latest BETA versions of BlueOS should support the latest ubuntu. We are also releasing 64bits images. E.g: https://github.com/bluerobotics/BlueOS/releases/tag/1.4.0-beta.7

Ryanf55, 2025-09-13:
> Hi, the latest BETA versions of BlueOS should support the latest ubuntu. We are also releasing 64bits images. E.g: https://github.com/bluerobotics/BlueOS/releases/tag/1.4.0-beta.7

Can you update the documentation? 
https://blueos.cloud/docs/stable/integrations/hardware/required/onboard-computer/

I'd be keen to test this on Nvidia Jetson hardware.

patrickelectric, 2026-01-14:
We have updated the README with instructions to run BlueOS:
https://github.com/bluerobotics/BlueOS#custom-installation