bluerobotics/BlueOS #1073: install: remove dependence on `pi` user

ES-Alexander, 2022-07-13:
Raised in #1072, and relevant because:
- not every OS has `pi` as the default user (e.g. Ubuntu)
- support for non-Raspberry Pi boards will likely require support for alternative user names

Should either:
- change script to use `$USER` or `$USERNAME` instead of `pi`, or
- set a `target_user` variable (or similar) at the top which defaults to `pi` (or `$USER`) and allows specifying something else with a command-line argument

Jollerprutt, 2022-09-04:
The Raspberry Pi foundation has [announced](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/) they are dropping pi as the default user, and that there will no longer be any default username.
So this issue will soon become more urgent as people start using the newer images.

ES-Alexander, 2023-06-16:
Will also need to change [commonwealth](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/libs/commonwealth/commonwealth/utils/commands.py#L5)

ES-Alexander, 2025-10-02:
> support for non-Raspberry Pi boards...

[Case in point](https://discuss.bluerobotics.com/t/jetson-nano-as-onboard-computer/14016/20).