bluerobotics/BlueOS #1093: core: frontend: Endpoints page selection options

ES-Alexander, 2022-07-26:
### Current behaviour

When creating a new endpoint you have to specify an IP and port for TCP/UDP or a device path and baudrate for serial, all via raw text. That's unintuitive, and relies on users knowing more than they should really have to when trying to achieve a simple integration/connection.

### Expected or desired behaviour

There should be context-aware dropdowns (based on the connection type), whereby

IPs can be selected from:
- Internal Link (loopback - `127.0.0.1`)
- External Link (localhost - `0.0.0.0`)
- Custom (allows custom IP, perhaps ideally as 4 number fields separated by dots?)

Serial devices can be selected from 
- Available devices (with relevant device information, like on the Bridges page), or 
- A custom path (perhaps needs to be chosen from at least one of the existing paths, even if it's not currently connected to a device)

Ports can be selected from:
- An open service (for internal links)
- Possibly something related to #470 (assuming an active port can be connected to)
- Custom

Baudrates can be selected from:
- A serial default (like already selectable on the Bridges page), or
- Custom 

---
Supporting "Custom" could either use a value in each dropdown, or there could be a separate "custom"/"development" checkbox that just swaps out the dropdowns with text input boxes.

---
Given both pages are about serial and UDP/TCP connections, does it maybe make sense to merge the Bridges and Endpoints pages into a single "Linking" page or something?

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-07-26:
We should definitely improve that page, and I think it's exactly around what you're suggesting.

rafaellehmkuhl, 2022-09-01:
@Williangalvani you've solved those right?

ES-Alexander, 2022-09-02:
@rafaellehmkuhl it's definitely improved with default reasonable IPs for the connection type, but there's no port selection, and for serial connections there's a default to one of the available devices but no device information and no way of selecting other devices, as well as no help with baudrate selection.

Accordingly I wouldn't call this issue solved yet, but it has been worked towards :-)