bluerobotics/BlueOS #1097: wifi-manager: Service (both hotspot and wifi client) does not work for newly-flashed-images

rafaellehmkuhl, 2022-08-01:
Daniel (from BR's support) reported that features deployed with #1019 worked perfectly with a docker-image-update from Beta4 to Beta5, but when he flashed an image with Beta5 directly, both the hotspot and the wifi client were not working.

From tests, I recognized that a just-flashed raspbian image has its wlan interface not configured, having most of its channels disabled. For the wifi-client, this is not a problem, but for the hotspot, this makes it not work properly. Since the `hostapd` fails to start, it ended up killing the wifi-manager service entirely.

We need to add considerations in the code for the possibility of the `hostapd` to fail to start, as well as doing something about the not-configured wlan interface.

rafaellehmkuhl, 2022-08-01:
Not completely solved. If there are no requests in the interface, it doesn't perform the initial configuration.

In my initial tests, those initial requests were being performed because there was an open BlueOS tab working over ethernet, so the wifi service was performing requests, and so the interface got configured. In the tests with the new image (PR #1096 merged) the interface was not configured till the time I've opened a tab and the first request was made.

patrickelectric, 2022-08-01:
nice find!

PR #1098 (2022-08-01): wifi-manager: Perform initial scan by default
When a raspbian image is flashed, the wlan interface is not configured until it receives an initial request. If the interface is not configured, the hotspot does not work correctly. To force the initial configuration, we perform an initial scan.

Tested with a freshy-flashed image. Now working flawlessly.

Fix #1097
closed 2022-08-01 by patrickelectric