bluerobotics/BlueOS #1115: Linux2rest is peaking at 75% cpu usage intermitently

Williangalvani, 2022-08-09:
The source of the issue seems to be a service in blueOS consuming all endpoints in short succession.

I suspect that [Helper is probing all endpoints assuming it could be a versioned API](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/helper/main.py#L88-L93)

ES-Alexander, 2022-08-09:
> **ES-Alexander:**
> Is there a general way to fix that, or do we need to create a blacklist file or something?

> **Williangalvani:**
> I have to read it more carefully, maybe some validation will suffice
> say use a regex to check if the link looks like a version name

patrickelectric, 2022-08-09:
Is there a log or something that can help us to debug it ?
Can you also provide the steps to replicate ? I got no over than 40% here with multiple BlueOS pages open in my computer.

Williangalvani, 2022-08-09:
> Is there a log or something that can help us to debug it ? Can you also provide the steps to replicate ? I got no over than 40% here with multiple BlueOS pages open in my computer.

the requests are done from the backend itself, which is why I think it is the helper service.
if you run linux2rest in verbose mode, you can see these requests come all at once, and the user agent is "python/requests"

Williangalvani, 2022-08-09:
[image]

patrickelectric, 2022-08-09:
Was able to replicate
[image]
<details>
  <summary>5min log:</summary>

INFO  12:55:48.687 src/logger/mod.rs:43: linux2rest, version: 0.4.5-e6285af, build date: 2022-08-04
INFO  12:55:48.687 src/logger/mod.rs:50: Starting at 2022-08-09T12:55:48
DEBUG 12:55:48.687 src/logger/mod.rs:54: Configuration: Arguments {
    verbose: true,
    port: 6030,
}
INFO  12:55:48.689 /root/.cargo/registry/src/github.com-1285ae84e5963aae/actix-server-1.0.4/src/builder.rs:263: Starting 4 workers
INFO  12:55:48.691 /root/.cargo/registry/src/github.com-1285ae84e5963aae/actix-server-1.0.4/src/builder.rs:277: Starting "actix-web-service-0.0.0.0:6030" service on 0.0.0.0:6030
DEBUG 12:55:51.736 src/server/pages.rs:79: 
HttpRequest HTTP/1.0 GET:/serial
  query: ?"udev=true"
  headers:
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"
, SerialQuery {
    udev: Some(
        true,
    ),
}
DEBUG 12:55:53.534 src/server/pages.rs:173: 
HttpRequest HTTP/1.0 GET:/platform
  headers:
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"

DEBUG 12:55:57.043 src/server/pages.rs:79: 
HttpRequest HTTP/1.0 GET:/serial
  query: ?"udev=true"
  headers:
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"
, SerialQuery {
    udev: Some(
        true,
    ),
}
DEBUG 12:55:58.543 src/server/pages.rs:173: 
HttpRequest HTTP/1.0 GET:/platform
  headers:
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"

DEBUG 12:56:02.371 src/server/pages.rs:79: 
HttpRequest HTTP/1.0 GET:/serial
  query: ?"udev=true"
[... 1858 lines cut ...]
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"
, SerialQuery {
    udev: Some(
        true,
    ),
}
DEBUG 13:00:25.964 src/server/pages.rs:173: 
HttpRequest HTTP/1.0 GET:/platform
  headers:
    "accept": "application/json, text/plain, */*"
    "accept-encoding": "gzip, deflate"
    "host": "127.0.0.1:6030"
    "user-agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36"
    "referer": "http://blueos.local/tools/web-terminal"
    "connection": "Upgrade"
    "accept-language": "pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6,ja;q=0.5"

INFO  13:00:26.745 /root/.cargo/registry/src/github.com-1285ae84e5963aae/actix-server-1.0.4/src/builder.rs:322: SIGINT received, exiting
</details>

PR #1119 (2022-08-10): core: services: helper: Cache detect_service output
Fix #1115

- There is no reason to not cache it without timeout.
- There is no way for us to check where is the swagger website without probing the valid paths.
closed 2022-08-10 by patrickelectric