bluerobotics/BlueOS #1123: ardupilot-manager: can't get available firmwares because of SSL error

rafaellehmkuhl, 2022-08-18:
Tracking for [this problem in our forums](https://discuss.bluerobotics.com/t/ardusub-firmware-update-no-data-available/12589).

This is the relevant log:
```bash
2022-08-12 22:58:51.715 | ERROR    | commonwealth.utils.apis:custom_route_handler:34 - Unhandled service exception.
2022-08-12 22:58:51.716 | ERROR    | commonwealth.utils.apis:custom_route_handler:35 - <urlopen error [SSL] unknown error (_ssl.c:1129)>
Traceback (most recent call last):

  File "/usr/local/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
    │ │       │   │             │   │         │   │     └ {'Host': 'firmware.ardupilot.org', 'User-Agent': 'Python-urllib/3.9', 'Connection': 'close'}
    │ │       │   │             │   │         │   └ <property object at 0xb5affac8>
    │ │       │   │             │   │         └ <urllib.request.Request object at 0xb4de2af0>
    │ │       │   │             │   └ '/manifest.json.gz'
    │ │       │   │             └ <urllib.request.Request object at 0xb4de2af0>
    │ │       │   └ <function Request.get_method at 0xb5b0a9b8>
    │ │       └ <urllib.request.Request object at 0xb4de2af0>
    │ └ <function HTTPConnection.request at 0xb5e3a7c0>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
    │    │             │       │    │     │        └ False
    │    │             │       │    │     └ {'Host': 'firmware.ardupilot.org', 'User-Agent': 'Python-urllib/3.9', 'Connection': 'close'}
    │    │             │       │    └ None
    │    │             │       └ '/manifest.json.gz'
    │    │             └ 'GET'
    │    └ <function HTTPConnection._send_request at 0xb5e3a808>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
    │    │          │                    └ False
    │    │          └ None
    │    └ <function HTTPConnection.endheaders at 0xb5e3a778>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
    │    │            │                            └ False
    │    │            └ None
    │    └ <function HTTPConnection._send_output at 0xb5e3a580>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
    │    │    └ b'GET /manifest.json.gz HTTP/1.1\r\nAccept-Encoding: identity\r\nHost: firmware.ardupilot.org\r\nUser-Agent: Python-urllib/3....
    │    └ <function HTTPConnection.send at 0xb5e3a4a8>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 980, in send
    self.connect()
    │    └ <function HTTPSConnection.connect at 0xb5e3a8e0>
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/http/client.py", line 1454, in connect
    self.sock = self._context.wrap_socket(self.sock,
    │    │      │    │        │           │    └ None
    │    │      │    │        │           └ <http.client.HTTPSConnection object at 0xb4de21c0>
    │    │      │    │        └ <function SSLContext.wrap_socket at 0xb65494a8>
    │    │      │    └ <ssl.SSLContext object at 0xb5624ec8>
    │    │      └ <http.client.HTTPSConnection object at 0xb4de21c0>
    │    └ None
    └ <http.client.HTTPSConnection object at 0xb4de21c0>
  File "/usr/local/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
           │    │               └ <classmethod object at 0xb653ff28>
           │    └ <class 'ssl.SSLSocket'>
           └ <ssl.SSLContext object at 0xb5624ec8>
  File "/usr/local/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
    │    └ <function SSLSocket.do_handshake at 0xb654bbf8>
    └ <ssl.SSLSocket [closed] fd=-1, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6>
  File "/usr/local/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
    │    └ None
[... 98 lines cut ...]
             │    └ <function OpenerDirector._call_chain at 0xb5b0ad60>
             └ <urllib.request.OpenerDirector object at 0xb56304f0>
  File "/usr/local/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
             │     └ (<urllib.request.Request object at 0xb4de2af0>,)
             └ <bound method HTTPSHandler.https_open of <urllib.request.HTTPSHandler object at 0xb4de2298>>
  File "/usr/local/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
           │    │       │    │      │                └ <urllib.request.Request object at 0xb4de2af0>
           │    │       │    │      └ <class 'http.client.HTTPSConnection'>
           │    │       │    └ <module 'urllib3.packages.six.moves.http_client' (<urllib3.packages.six._SixMetaPathImporter object at 0xb5bf73b8>)>
           │    │       └ <module 'http' from '/usr/local/lib/python3.9/http/__init__.py'>
           │    └ <function AbstractHTTPHandler.do_open at 0xb5b0ead8>
           └ <urllib.request.HTTPSHandler object at 0xb4de2298>
  File "/usr/local/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
          └ <class 'urllib.error.URLError'>

urllib.error.URLError: <urlopen error [SSL] unknown error (_ssl.c:1129)>
2022-08-12 22:58:51.822 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:54494 - "GET /v1.0/available_firmwares?vehicle=Sub HTTP/1.0" 500
```

The problem is probably related to [this line](https://github.com/bluerobotics/BlueOS-docker/blob/5edf44be291608aab29a590f115f40c79b181043/core/services/ardupilot_manager/firmware/FirmwareDownload.py#L28).

The comment [in this answer](https://stackoverflow.com/a/52172355/4465365) probably solves our problem.

patrickelectric, 2022-11-14:
@rafaellehmkuhl what is the status of this issue ?

rafaellehmkuhl, 2022-11-15:
> @rafaellehmkuhl what is the status of this issue ?

Good question. I'm pretty sure it's unsolved.

patrickelectric, 2024-06-19:
From our tests, it appears that we cant replicate it anymore.

closed 2024-06-19 by patrickelectric