bluerobotics/BlueOS #1150: bug: network test upload speed not consistently displaying

ES-Alexander, 2022-09-16:
### Bug description

When running the network test, the upload speed sometimes seemingly gets measured but then doesn't display in the frontend.

Raised in [this forum post](https://discuss.bluerobotics.com/t/can-not-test-the-upload-speed/12900), which provided the following browser console log:
[192.168.2.2-1663247214139.log](https://github.com/bluerobotics/BlueOS-docker/files/9582359/192.168.2.2-1663247214139.log)

### Steps to reproduce

1. Go to the [Network Test](https://docs.bluerobotics.com/ardusub-zola/software/onboard/BlueOS-1.0/advanced-usage/#network-test) page
2. Run the test
3. Upload speed may display as `...` instead of a measured number

### Primary pain point(s)

Doesn't stop the system from working, but makes troubleshooting harder due to not knowing what kind of network bandwidth is available.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2022-09-16:
Discussed internally [here](https://bluerobotics.slack.com/archives/C02UF32NXN0/p1663240073019279).

@joaoantoniocardoso was able to replicate the issue, and has said
> We've identified the problem, it will be fixed on the next release :)

joaoantoniocardoso, 2022-09-16:
Thanks, it was fixed here: https://github.com/bluerobotics/BlueOS-docker/pull/1149, and it is already available on the current master.

closed 2022-09-16 by ES-Alexander