bluerobotics/BlueOS #1152: ardupilot_manager: hide unsupported firmwares

ES-Alexander, 2022-09-20:
### Current behaviour

There are firmware builds for boards that are known to not fully work, but the firmware manager still shows all of them.

### Expected or desired behaviour

If 'bad' versions are known, hide them. In particular it would be good to support minimum versions (e.g. Navigator requires Sub >= 4.1.0, and Copter/Plane/Rover >= 4.2.3; and Pixhawk 4 also requires Sub >= 4.1.0).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-09-20:
Maybe we should put an icon at the side of the firmware version with a tooltip saying this version is known to present problems? Maybe just for pirate mode and actually hide in regular mode?

I ask this because:
1. this list is going to be handcrafted
2. we may still want to see and test those firmwares
3. maybe we will blacklist version that users actually want to use, be for tests or whatever

ES-Alexander, 2022-09-22:
In general that probably makes sense :-)
I was thinking of this mostly as versions where we can say "this is broken enough that people will almost certainly not want to use it" should be hidden by default (which is mostly relevant for minimum versions for a given hardware, but could also happen if a new very broken version somehow gets released).

patrickelectric, 2024-06-19:
I believe that @Williangalvani will be working on that, using the current information in our parameter repository folder.