bluerobotics/BlueOS #1153: bug: ardupilot-manager: ArduCopter failing to fetch for Navigator

ES-Alexander, 2022-09-20:
### Bug description

Navigator firmwares are fetching fine, except for Copter. The https://firmware.ardupilot.org/Copter/stable-4.2.3/navigator/ webpage loads fine, so not sure why the fetch is failing.

[logfile_09-20-2022_00_34_19.log](https://github.com/bluerobotics/BlueOS-docker/files/9603115/logfile_09-20-2022_00_34_19.log)

### Steps to reproduce

1. Connect Navigator
2. Open General page
3. Try to fetch Copter firmware -> fails
4. Try to fetch other vehicle type (Sub/Plane/Rover) -> succeeds

### Primary pain point(s)

Requires people wanting to use ArduCopter to do a manual upload, which is more complicated and less intuitive than the normal simple installation process.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-10-17:
This is happening because [there are multiple candidates for download](https://firmware.ardupilot.org/Copter/beta/),`navigator` and `navigator-heli`, [different from Sub](https://firmware.ardupilot.org/Sub/beta/), for example.

[image]

I will work on a workaround for this.

closed 2022-10-17 by patrickelectric

rafaellehmkuhl, 2022-10-18:
[image]

Proof of fix.

PR #1181 (2022-10-17): ardupilot-manager: Send first found firmware candidate when multiple exist
Current behavior was to raise, but that prevent firmware to be downloaded.

Core team agreed it's better to have one than to have nothing, so we go on with the first candidate and log a warning message about the problem.

This is experienced for ArduCopter firmware, since it has both "regular" and "heli" firmware available in the same repo.

Image building and testing is in progress.

Fix #1153