bluerobotics/BlueOS #1156: bug: mavlink-router fails to create multiple TCP servers

flaeri, 2022-09-23:
### Bug description

In the web UI it seems perfectly ok to attempt to create more than one TCP server, say with different ports. This is something I've been accustomed to prior to the docker implementation, and has been very useful for us.

In the current (docker) solution, it would seem like its working, but it is not.
Only one of the ports would spawn (seemingly at random). Sometimes there would exist more listen ports, but they would quickly die (disappear).

### Steps to reproduce

1. Go to the WebUI (http://x.x.x.x/autopilot/endpoints)
2. Attempt to create two or more TCP-server endpoints (for instance 0.0.0.0:14660 + 0.0.0.0:14550)
[image]
3. Observe only one of the ports exist using `netstat -tulnp`, and confirm you are unable to access the listen ports you defined

### Primary pain point(s)

When using multiple computer and/or systems (for instance, self developed python scripts + QGrondControl), re-using the same listen port causes problems (one would seem to be blocking the other).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-09-23:
If I remember correctly, we don't do that because `mavlink-routerd` does not support multiple TCP servers working simultaneously. @Williangalvani @joaoantoniocardoso do you remember that?

flaeri, 2022-09-23:
I see, that is a bit of a bummer for our project, hehe. If its not a feasible change, then perhaps there could be some sort of UI check in place to guard against this behavior.

Hope its not too poor taste to sneak in a little question about alternatives. Could I for instance have mavlink-routerd pass all the packets to a mavproxy running locally on the same network?

joaoantoniocardoso, 2022-09-23:
> If I remember correctly, we don't do that because `mavlink-routerd` does not support multiple TCP servers working simultaneously. @Williangalvani @joaoantoniocardoso do you remember that?

Yes, i can confirm that `mavlink-routerd` is limited to only one TPC server, and this is one of our main concerns about continuing using it.

@flaeri Maybe you could try to use the already running TCP server port (since _BlueOS beta 9_) that is being used for the mavlink-camera-manager - `5777`.

If that doesn't work for you, then the only solution I can think of would be to migrate from `mavlink-routerd` to a more capable one, like [mavp2p](https://github.com/aler9/mavp2p). To do so, probably our `ardupilot_manager` service would need to be changed to support `mavp2p`.

rafaellehmkuhl, 2022-09-23:
> I see, that is a bit of a bummer for our project, hehe. If its not a feasible change, then perhaps there could be some sort of UI check in place to guard against this behavior.
>
> Hope its not too [...]

The alternative you've mentioned is possible. You can just chain `mavproxy` on the existing `mavlinkrouterd` and from there use `mavproxy` capabilities.

As @joaoantoniocardoso mentioned, we are planning to change `mavlinkrouterd` for something else in the future, since there are many limitations on it that are preventing us from improve mavlink routing capabilities.

flaeri, 2022-09-24:
> The alternative you've mentioned is possible. You can just chain `mavproxy` on the existing `mavlinkrouterd` and from there use `mavproxy` capabilities.

@rafaellehmkuhl and I would do that via normal means in mavlink-routerd, and have the mavproxy "master" pointed/listening for those packets?

Excited to see what the future brings. Thank you all very much for all your hard work, and for your insights!

rafaellehmkuhl, 2022-09-25:
> > The alternative you've mentioned is possible. You can just chain `mavproxy` on the existing `mavlinkrouterd` and from there use `mavproxy` capabilities.
> 
> @rafaellehmkuhl and I would do that via normal means in mavlink-routerd, and have the mavproxy "master" pointed/listening for those packets?

Exactly that. While we don't yet support extensions, you could put `mavproxy` at the Host OS (RaspBian) on your companion computer.
 
> Excited to see what the future brings. Thank you all very much for all your hard work, and for your insights!

We are also very excited with what we are going to deliver in the next months! A lot of new and cool stuff that we think the community will love. Thanks!