bluerobotics/BlueOS #1200: bug: Ping service: Is creating pings with same port

patrickelectric, 2022-10-26:
### Bug description

[image]

### Steps to reproduce

Connect two pings and a ping360

### Primary pain point(s)

The UI does not make sense and it's not possible to know which sensor is being used 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2022-10-26:
that is interesting. I wonder if bridges is stopping and that is causing the port to be identified as unused.

Can you share the log file?

patrickelectric, 2022-10-27:
It was a report from manufacture, they are going to get in touch

ES-Alexander, 2023-06-19:
Also raised in [this forum post](https://discuss.bluerobotics.com/t/ping-sonar-altimeter-udp-port-overlap-in-blue-os/14506) - I've asked for a log.

Since it's async, I suspect the driver is likely starting to detect the next device before the previous device is counted as registered in the OS. @Williangalvani not sure if you've got some idea how to fix/avoid that.

ES-Alexander, 2024-01-29:
This has been [raised again](https://discuss.bluerobotics.com/t/ping-sonar-altimeter-udp-port-overlap-in-blue-os/14506/5), but I've also realised a possible cause, and thought of some ideas for potentially fixing it:

>... my suspicion is that the asynchronous port searching means both device connections find the same UDP port before either of them is fully connected to it (in a race condition), and then they end up sharing the connection.
>
>That may be preventable with a well-placed sleep, or perhaps by making the port connection process raise an error if multiple connections are made to the same port, and/or treatable by adding duplicate-port detection to the ping service and making it restart relevant device connections when port sharing is detected.

ES-Alexander, 2024-01-29:
It may also be worth just storing a set of known used ports as a blacklist, since I imagine Python's variables will update faster than a port connection does. It's more important to have unique ports than to always have the same / consistent ports, so if we use a blacklist + port connection checking then that may be sufficient.

PR #2354 (2024-01-29): core: services: ping: minimise port duplication races
Hopefully fixes #1200 (or at least significantly reduces the likelihood of it).

**Details:**
- ⚠️ Untested
    - should be testable at `esalexander/blueos-core`, `:ping-fix-attempt` (in the BlueOS Version Chooser)
- Uses the approach I suggested [here](https://github.com/bluerobotics/BlueOS/issues/1200#issuecomment-1914132647)
- It would be fully robust to also/instead regularly check for duplicated ports in the drivers (and restart them as necessary), but that seems excessive to me, and more involved to program and test
- Ideally this cod [... cut]
closed 2024-01-29 by Williangalvani