bluerobotics/BlueOS #1205: bug: install.sh on Raspberry Pi requires Bullseye otherwise errors

matt-bathyscope, 2022-11-02:
### Bug description

The `install.sh` script assumes your Raspberry Pi is already on a Debian Bullseye variant of Raspberry Pi OS. This may not be the case if you're upgrading an older companion. The `get_legacy` function of `raspi-config` appears to only be present on the Bullseye variants which causes an error on Buster variants.

```
/usr/bin/raspi-config
Running in a Raspiberry.
/usr/bin/raspi-config: 2569: /usr/bin/raspi-config: get_legacy: not found
```

If the intention is to have everyone on Bullseye then maybe fail sooner with a message to update. If Buster is meant to still be supported, I recommend gating the `raspi-config nonint get_legacy` command (https://github.com/bluerobotics/BlueOS-docker/blob/master/install/install.sh#L168) with a check to see if you're actually on Bullseye. This compatibility step is not needed on Buster.

I'm happy to PR this to either fail early if you're not on Bullseye, or add a check `lsb_release -sc` is `bullseye` before the camera compatibility command is run.

### Steps to reproduce

Run the install script per README

```
sudo su -c 'curl -fsSL https://raw.githubusercontent.com/bluerobotics/blueos-docker/master/install/install.sh | bash'
```

### Primary pain point(s)

Installer script doesn't succeed on Raspberry Pi OS unless you're on Bullseye.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2022-11-02:
Thanks for the report @matt-bathyscope!

matt-bathyscope, 2022-11-02:
Is the idea to require Bullseye?

patrickelectric, 2022-11-03:
> Is the idea to require Bullseye?

bullseye is not required, but if I recall correctly, is the last version that still has support to the raspberry legacy mode. 
That's necessary since the new mode does not have official support from gstreamer, making it not possible for us to stream. 
So, if you plan to use only USB cameras and not the CSI Raspberry camera, you should be fine.