bluerobotics/BlueOS #1245: Display only parameters edited by user

patrickelectric, 2022-11-30:
It should be nice to have a checkbox where only edited parameters would appear, showing the new value and the default one

Williangalvani, 2022-11-30:
I wonder if we can get this info via mavlink

ES-Alexander, 2022-12-01:
That's a really cool idea, and would likely be helpful for support :D

@Williangalvani if it's not possible to get directly via MAVLink, presumably this could be done by comparing the values in the default parameter file (from either BlueOS or stored in the firmware) with the current parameters?

I know there are some parameters that don't get changed by setting new defaults, but I'm not sure of the mechanism for that. Are those parameters just not included in the defaults file (in which case it wouldn't help with finding if they've been changed), or are they included but just not applied somehow?

patrickelectric, 2022-12-01:
From what I could search, it's not possible to get the default parameters value from the firmware using the PARAM architecture.

`AP_GROUPEND` is a generic `none` parameter and the parameters stored inside `::var_info`does not have a common header. Besides that, all the `AP_Param` macros don't have a lookup value that is simpler to find inside the firmware binary, the parameters names are also not stored entirely but as index. 

From what I can find, the only way of getting the default values is to fetch it when the firmware run for the first time on the ardupilot manager side.

patrickelectric, 2022-12-01:
It may be possible with the firmware binary, while using the param file to have the name of all parameters, we can do a look back of the address of such names on the binary and based on it grab the default values, doable but complex.

patrickelectric, 2024-06-19:
That's already done.

closed 2024-06-19 by patrickelectric