bluerobotics/BlueOS #1260: bug: kernel panic after running installation script

jaxxzer, 2022-12-08:
### Bug description

cmdline.txt is corrupted after running installation script, resulting in kernel panic at boot.

The content after install is:
`console=tty1 root=PARTUUID=2985b382-02 rootfstype=ext4 fsck.repair=yes rootwai cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory`

The pi boots after fixing the word 'rootwait'

### Steps to reproduce

Install Raspberry Pi OS Lite (32 or 64 bit)
ssh to raspberry pi and run `sudo su -c 'curl -fsSL https://raw.githubusercontent.com/bluerobotics/blueos-docker/master/install/install.sh | bash'`
pi will kernel panic on boot

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2022-12-08:
maybe [this](https://github.com/bluerobotics/BlueOS-docker/blob/master/install/boards/bcm_27xx.sh#L73) was written to get rid of a newline and is now removing the last characater? maybe raspbian removed the newline on the new releases? just a guess.

jaxxzer, 2022-12-08:
[image]

patrickelectric, 2022-12-08:
Able to replicate with the described steps

jaxxzer, 2022-12-08:
The content of cmdline.txt after imaging the card (before booting) is:
`console=serial0,115200 console=tty1 root=PARTUUID=21e60f8c-02 rootfstype=ext4 fsck.repair=yes rootwait quiet init=/usr/lib/raspberrypi-sys-mods/firstboot systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target`

No newline

jaxxzer, 2022-12-08:
The content of cmdline.txt after first boot is:

`console=serial0,115200 console=tty1 root=PARTUUID=1676c83a-02 rootfstype=ext4 fsck.repair=yes rootwait`

The content of cmdline.txt after running @patrickelectric command:

`console=tty1 root=PARTUUID=1676c83a-02 rootfstype=ext4 fsck.repair=yes rootwait cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory`

and no more kernel panic after reboot

PR #1261 (2022-12-08): install: boards: Use sed over truncate to add cgroups
Truncate can remove characters of files that does not have a line break.
Sed can add the text on the first line without problems.

Fix #1260
closed 2022-12-08 by patrickelectric