bluerobotics/BlueOS #1268: core: ardupilot_manager: add support for aarch64 ardupilot binaries on aarch64 installs

jaxxzer, 2022-12-09:
### Current behaviour

- Install BlueOS on Raspberry Pi OS Lite 64 bit 2022-09-22
- try to upload the firmware file: [ardusub_aarch64.zip](https://github.com/bluerobotics/BlueOS-docker/files/10191645/ardusub_aarch64.zip)
- check fails:

[image]

### Expected or desired behaviour

It goes!

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2022-12-09:
Can you use the 32 bits binary ?

patrickelectric, 2022-12-09:
@jaxxzer check the latest master version available, we updated the library that could contain a problem. [heads/master-0-g2088489](https://github.com/bluerobotics/blueos-docker/tree/2088489)

jaxxzer, 2022-12-09:
I get the same result:
[image]

supplemental:
```
[jacob@jacob-laptop Downloads]$ readelf -h ardusub_aarch64
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           AArch64
  Version:                           0x1
  Entry point address:               0x4048cc
  Start of program headers:          64 (bytes into file)
  Start of section headers:          2754512 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         37
  Section header string table index: 36
```

```
[jacob@jacob-laptop Downloads]$ pip freeze | grep pyelftools
pyelftools==0.29
[jacob@jacob-laptop Downloads]$ python
Python 3.10.6 (main, Aug  3 2022, 17:39:45) [GCC 12.1.1 20220730] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.machine()
'x86_64'
>>> from elftools.elf.elffile import ELFFile
>>> with open ("ardusub_aarch64", "rb") as f:
...     elf = ELFFile(f)
...     print(elf.get_machine_arch())
... 
AArch64
>>> 
```

[image]

matt-bathyscope, 2022-12-12:
This was also raised in forum feedback [here](https://discuss.bluerobotics.com/t/navigator-firmware-for-aarch64/12412) and [here](https://discuss.bluerobotics.com/t/64-bit-os-and-containers/11509).

Changing [this dictionary](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/ardupilot_manager/firmware/FirmwareInstall.py#L28) to add `"aarch64": "AArch64"` will allow the ELF image to validate and install, but it's not actually successful because the core Docker image is 32 bit and won't be able to execute the 64 bit binary.

I think the right option here is to move entirely to `aarch64` for the containers and the host OS. This would also require shipping an `aarch64` build of ArduSub for Navigator - which seemingly requires changes to how firmware binaries are discovered and downloaded since I think everything in the ArduPilot manifest is 32 bit, but I'm not sure. Spot checking the Navigator board config shows it has `toolchain = "arm-linux-gnueabihf"` set [here](https://github.com/ArduPilot/ardupilot/blob/99f6a3118893b2f6e68fb05a2d9171fb022e290a/Tools/ardupilotwaf/boards.py#L1145) which will only produce a 32 bit binary.

I'm open to helping with this because it simplifies build and delivery for our integration, but I think more discussion/planning is needed here since there are so many dependencies.

Williangalvani, 2022-12-16:
Hi @matt-bathyscope ,

Do you really need a 32bit binary? wont the 32 bit one run in the 64 bits OS?

matt-bathyscope, 2022-12-16:
I think the main issue is the Docker container's OS has no loader for `aarch64` binaries. Statically linked binaries will execute fine, but there is no `aarch64` loader in the `armv7l` Docker image. For example:

```
root@blueos:~/.config/ardupilot-manager/firmware# file ardupilot_navigator 
ardupilot_navigator: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=24c81b967ecf6a771144608dd95a1a2d3fe131e7, not stripped

root@blueos:~/.config/ardupilot-manager/firmware# file ardusub_aarch64 
ardusub_aarch64: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, with debug_info, not stripped

root@blueos:~/.config/ardupilot-manager/firmware# file deepside2_linux_aarch64 
deepside2_linux_aarch64: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, Go BuildID=fT8Y65qVdJBiEQR-8AT9/GE6w8z0LIhMDfRQ4_Igl/zvCkEKZSqytHkcU55hQa/h3u4EkYFIqcQn0bH_nHD, with debug_info, not stripped

root@blueos:~/.config/ardupilot-manager/firmware# ls -la /lib/ld-linux-armhf.so.3
lrwxrwxrwx 30 root 14 Oct 19:35 /lib/ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.31.so

root@blueos:~/.config/ardupilot-manager/firmware# ls -la /lib/ld-linux-aarch64.so.1
"/lib/ld-linux-aarch64.so.1": No such file or directory (os error 2)

root@blueos:~/.config/ardupilot-manager/firmware# ./ardusub_aarch64 
bash: ./ardusub_aarch64: No such file or directory

root@blueos:~/.config/ardupilot-manager/firmware# ./deepside2_linux_aarch64 
2022/12/16 18:56:03 Bathyscope Deepside2 
2022/12/16 18:56:03 **** RUNNING IN DEBUG MODE ****
2022/12/16 18:56:03 Is Raspberry Pi: true
2022/12/16 18:56:03 Topside UUID: <snipped>
```

You can see the loader for the `ardusub_aarch64` binary doesn't exist and when you attempt to execute that binary you get a `No such file or directory` error, whereas our binary (which is statically linked) will execute in the container.

It may be possible to add `aarch64` as a foreign architecture and bring in all the libraries needed to dynamically link dependencies, but I don't think that makes as much sense as moving to `aarch64` natively.

Williangalvani, 2023-01-26:
in order to get this done we need to:
- [ ] have a v8 build of our base docker (I just opened a pr)
- [ ] have a v8 build of core
- [ ] optional - update our image building system to armv8

I do expect we'll run into some issues along the way, though.

ES-Alexander, 2023-10-11:
Raised again in [this forum post](https://discuss.bluerobotics.com/t/autopilot-manager-vehicle-type-fetch-fail/15175).

closed 2023-11-22 by patrickelectric

matt-bathyscope, 2023-11-22:
Nice!

PR #2174 (2023-11-22): Add initial support to aarch64 firmware 
Fix #1268