bluerobotics/BlueOS #1296: Add webpage/system telemetry to BlueOS

patrickelectric, 2023-01-03:
Right now we don't have information about our users besides the surveys in our forum.
Adding telemetry in BlueOS could automatically inform us about a series of key factors, such as: BlueOS version, vehicle type, number of missions and more.

Williangalvani, 2023-02-23:
What do we want:
 - Basic system information:
   - Host Hardware
     - Sd card size/storage status
     - Detected peripherals
     - memory size
   - Ardupilot firmware
   - Cpu temperature Peak
 - BlueOS uptime
 - Ardupilot uptime
 - BlueOS  version
 
 Good to have:
   - Website interactions

patrickelectric, 2023-02-27:
* Water temperature (thruster design)
* Maximum and medium depth (products)

patrickelectric, 2025-01-11:
@voorloopnul what is the status ?