bluerobotics/BlueOS #1297: core: frontend: stop sidebar names from being truncated

ES-Alexander, 2023-01-04:
### Current behaviour

When the browser window gets shorter than the currently visible sidebar content a scrollbar is automatically created, but it takes width from the existing sidebar which ends up truncating the menu item names. That makes it harder to see what the options are, which is annoying for navigating the interface.

[image]

### Expected or desired behaviour

It's possible to work around the issue by shrinking the text or making the browser window taller, but that's not always reasonable to do. Some solutions in order of (my) preference:

1. the scrollbar could be added on to the width of the existing sidebar, instead of taking up already used width
1. the sidebar could be made to already include the width of a potential scrollbar
    - accordingly, if and when it appears it just takes its allocated space without affecting the other elements
    - this may look a bit messy because there'd be extra space when there's no scrollbar
1. the sidebar could have a couple of size options to choose from 
    - `[hidden, icons-only, icons+text]`
1. the sidebar could be made resizable via click+drag
    - this adds customisability, but I don't think it's necessary, and it also adds some cognitive load to the user

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2023-01-05:
I think the UI here could be a little different:
- remove the icons for the groups, and to give a clear separation of the groups, make them UPPERCASE and/or bold, but don't use a bigger font size than the rest, maybe even a slightly smaller one. If a better separation is needed, add a blank line before each group label
- keep icons for all items (inside or outside groups), but remove all that indentation;
- reduce the gap between icon and label;

IMHO it would look cleaner

ES-Alexander, 2023-01-23:
In internal discussion, @rafaellehmkuhl mentioned
> What I usually see for this is hiding the scrollbar and fading the bottom content, so people know there's more down and that they can scroll.

PR #1699 (2023-05-23): core: frontend: App: Turn project logo as clickable to go home
Rework side menu for the new style
Fix #1297 

new:

[image]

[image]
closed 2023-05-23 by patrickelectric