bluerobotics/BlueOS #1313: bug:  service running via tmux stops once scrolling up

patrickelectric, 2023-01-10:
### Bug description

Service stop once we use scroll 

### Steps to reproduce

Go to terminal 
Access tmux sessions
Choose one 
Scroll up
Service will stop running 

### Primary pain point(s)

The service stops running

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2023-01-11:
This is the normal behavior of tmux: [easy explanation](https://stackoverflow.com/a/14569349/3850957)

patrickelectric, 2023-01-11:
Moving away from tmux appears to be the proper solution to it, but I believe that is not a problem for the normal user, only for developers. We should be fine for now.

closed 2025-01-11 by patrickelectric