bluerobotics/BlueOS #1339: core: frontend: switch pirate mode via header icon

ES-Alexander, 2023-01-12:
### Current behaviour

Changing into and out of pirate mode requires going into the settings from the sidebar, which is inconvenient.

### Expected or desired behaviour

#1317 added a pirate icon to the header to indicate whether pirate mode is currently active. It would be nice to be able to just click on that icon to switch modes.

>> @patrickelectric:
>> how exactly do it, we are going to change the icon ? We are going to change the color ? What is the desired behavior of the UI
>
>> @ES-Alexander: 
>> I think changing icons is likely the most intuitive. Maybe swap between `pirate` (or `skull-crossbones`) and `emoticon-happy-outline` (or `emoticon-cool-outline`), with a tooltip that says "Click to {en/dis}able pirate mode."

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2023-01-12:
we could  also show a pop-up like we have for wifi, with a toggle-switch for on/off

ES-Alexander, 2023-01-13:
> we could also show a pop-up like we have for wifi, with a toggle-switch for on/off

We could, but given the icon and sidebar change, and generally some content on the page you’re on does too, it seems fine to me to save a click and a mouse move by just making the icon act as a toggle button.

PR #1372 (2023-01-18): New pirate mode UI
https://user-images.githubusercontent.com/1215497/212980960-e49c609c-fd97-4e83-bd15-4d6583d25f79.mp4

Fix #1339