bluerobotics/BlueOS #1416: bug: /boot/cmdline.txt not found

discoimp, 2023-01-30:
### Bug description

I'm running a Raspberry Pi 3b with Ubuntu 20.04 server.
The [bcm_28xx.sh](https://github.com/discoimp/BlueOS-docker_ubuntu20.04_server/blob/master/install/boards/bcm_28xx.sh) try to replace some lines in /boot/cmdline.txt, but on my OS the file is located at /boot/firmware/.

### Steps to reproduce

Simply run install.sh on a fresh install ubuntu 20.04 on a raspberry pi 3b:
Output:
```
Checking if network and remote are available: https://raw.githubusercontent.com/bluerobotics/blueos-docker/master
Starting hardware configuration.
Detecting board type
Detected BCM28XX via device tree
Configuring BCM28XX board (Raspberry Pi zero, 1, 2, 3)..
- Enable I2C, SPI and UART.
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=uart1
- Set up kernel modules.
bcm2835-v4l2
i2c-bcm2835
i2c-dev
- Configure serial.
sed: can't read /boot/cmdline.txt: No such file or directory
- Enable cgroup with memory and cpu
grep: /boot/cmdline.txt: No such file or directory
sed: can't read /boot/cmdline.txt: No such file or directory
```

### Primary pain point(s)

Something like this in line 30 of [bcm_28xx.sh](https://github.com/discoimp/BlueOS-docker_ubuntu20.04_server/blob/master/install/boards/bcm_28xx.sh) might work?:
```
# Remove any console serial configuration
echo "- Configure serial."
if [ ! -f "/boot/cmdline.txt"]; then
    echo "Creating symlink for cmdline.txt"
    ln -s /boot/firmware/cmdline.txt /boot/
fi
sudo sed -e 's/console=serial[0-9],[0-9]*\ //' -i $CMDLINE_FILE
```

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-01-31:
The script was created originally to work with the `cmdline.txt` provided by the official Raspberry image.

To make it work with ubuntu, I would recommend first to check if both OSs use the same `cmdline` format, and if that's the case, the only thing necessary is to set the `CMDLINE_FILE` on the beginning of the file, like:

```sh
if [ uname -a | grep -iq ubuntu ]; then
    CMDLINE_FILE=/boot/firmware/cmdline.txt
fi
```

Creating the `/boot/firmware/cmdline.txt` symlink may work, but it would be safer to check if the file exists first before creating it, including a comment explaining the reasons behind it.
The example that I provided already explain what is going on based on the code.

patrickelectric, 2024-05-13:
I'll close this for now, be free to reach us if you still have problems

closed 2024-05-13 by patrickelectric