bluerobotics/BlueOS #1418: Extension manager offers downgrade (to 0.0.0)

rafaellehmkuhl, 2023-01-30:
[image]

joaoantoniocardoso, 2023-01-30:
Also mentioned [here](https://github.com/bluerobotics/BlueOS-docker/issues/1398).

rafaellehmkuhl, 2023-02-01:
> I could not replicate this. any directions?

Really don't know.
I had just instaled latest BlueOS master and installed zerotier after that.
Maybe it's worth mentioning that I use to have zerotier before the update that removed all installed extensions.

joaoantoniocardoso, 2023-02-02:
> I could not replicate this. any directions?

[The video posted here](https://github.com/bluerobotics/BlueOS-docker/issues/1398) might give you some reproducible path

patrickelectric, 2023-02-03:
Is this fixed ? @Williangalvani

Williangalvani, 2023-02-03:
yes

closed 2023-02-03 by Williangalvani