bluerobotics/BlueOS #1444: core: support internet passthrough via tether

ES-Alexander, 2023-02-07:
### Current behaviour

Connecting the vehicle to the internet significantly simplifies updating and installing BlueOS versions, firmware, and extensions. Currently the only internet connectivity option for the vehicle is via wifi, which is problematic for users with metal enclosures and/or doing long-term deployments, whose only option is to manually download relevant files to their control computer (from the internet) and then upload them to the vehicle.

### Expected or desired behaviour

If the control computer has internet access, and a networked connection to the vehicle, it should be possible for the control computer to act as a proxy, and forward the internet access to the onboard computer via the existing connection.

This would significantly improve the user experience for all use-cases where providing wifi access to the onboard computer is challenging, and could also increase internet speeds to the vehicle if the control computer has a fast internet connection and a higher tether bandwidth than the onboard computer's wifi bandwidth capacity (which is likely to become more common as tether technologies improve).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2023-02-07:
I didn't find #1280 in my original search, but it's on the same issue. @patrickelectric's comment there at least addresses how to achieve passthrough Windows, but it would be good to have instructions for all supported OSs. From that link it seems like the actual functionality side of this is potentially entirely handled by the topside, in which case this is mostly a documentation issue, and the instructions would then be automatically included in BlueOS once we build the docs into it.

Not sure if BlueOS needs to do anything special for this to work, or if it would automatically recognise that its ethernet connection is providing internet access. It would at least be helpful for BlueOS to have some general way of showing that it has internet access, especially when that's not via wifi.

ES-Alexander, 2023-02-07:
Tried this [on Mac](https://support.apple.com/en-au/guide/mac-help/mchlp1540/mac) and it seemed to work fine (I was able to download and flash a new firmware). That said,

- Not having an indication that there was an internet connection (as per #1057) was very unintuitive to operate with
    - [These](https://github.com/bluerobotics/BlueOS-docker/pull/1305#pullrequestreview-1236888395) two [checks](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/frontend/src/views/MainView.vue#L120) being wifi only do not help, because they make it seem like there's no internet even if there is
    - It would be good if we could add an icon to the header specifically for internet connectivity (e.g. `mdi-web`), since neither a wifi not ethernet connection guarantees internet is available
- Adding internet testing (as per #443) would also be helpful for determining which connection is better in a vehicle that's capable of both

patrickelectric, 2023-06-21:
I think that we can close this issue since it's not connected to BlueOS and no BlueOS configuration is necessary to do it. Major OS on the surface computer already provides this functionality for BlueOS without hacking things around.

ES-Alexander, 2023-06-21:
The documentation side should be handled in our docs, but I'm also somewhat reluctant to document the process while #1057 and #448 aren't done, because it's not a great experience at the moment. Those issues already exist, so we _could_ close this one, but this is also a good reminder that might increase the motivation to get those done.

joaoantoniocardoso, 2023-06-22:
> I think that we can close this issue since it's not connected to BlueOS and no BlueOS configuration is necessary to do it. Major OS on the surface computer already provides this functionality for BlueOS without hacking things around.

I believe on BlueOS we'd need to do something simple like the ability to change the default route

ES-Alexander, 2023-12-04:
@joaoantoniocardoso I did :-)
bluerobotics/ardusub-zola#142