bluerobotics/BlueOS #1446: Allow bridging connections in the vehicle so that topside can access vehicle-side peripherals

Williangalvani, 2023-02-07:
use case in mind is setting up the Blue Boat networking (powerful radio connected to eth0) via blueos hotspot.

joaoantoniocardoso, 2023-02-07:
Not exactly the same thing, but might help - my computer had two network interfaces, one connected to the internet, and the other one connected to the vehicle. For me to share my internet to the vehicle I was using the following script:

```sh
#!/usr/bin/env sh

set -o verbose

INTERNET0=eno1
NET0=enp2s0f0u1

sudo ip link set up dev $INTERNET0
sudo ip link set up dev $NET0

sudo ip addr add 192.168.2.1/24 dev $NET0

sudo sysctl net.ipv4.ip_forward=1
sudo sysctl net.ipv6.conf.all.forwarding=1
sudo sysctl net.ipv6.conf.default.forwarding=1

sudo systemctl start iptables.service

sudo iptables -t nat -A POSTROUTING -o $INTERNET0 -j MASQUERADE
sudo iptables -A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i $NET0 -o $INTERNET0 -j ACCEPT
sudo iptables -I INPUT -p udp --dport 67 -i $NET0 -j ACCEPT
sudo iptables -I INPUT -p udp --dport 53 -s 192.168.2.0/24 -j ACCEPT
sudo iptables -I INPUT -p tcp --dport 53 -s 192.168.2.0/24 -j ACCEPT
sudo iptables -I INPUT -p udp --dport 8554 -s 192.168.2.0/24 -j ACCEPT
sudo iptables -I INPUT -p tcp --dport 8554 -s 192.168.2.0/24 -j ACCEPT

sudo iptables -L

```

ES-Alexander, 2023-02-13:
> For me to share my internet to the vehicle I was using ...

Potentially relevant to #1444

joaoantoniocardoso, 2023-02-14:
> > For me to share my internet to the vehicle I was using ...
> 
> Potentially relevant to #1444

I think it is exactly that :laughing:

patrickelectric, 2024-06-19:
Done by USBIP and VirtualHere BlueOS extensions

closed 2024-06-19 by patrickelectric

joaoantoniocardoso, 2024-06-20:
@patrickelectric I don't think this is the same think

Williangalvani, 2024-06-20:
I think these should be two different issues.
this one is also helped by the proxy extension