bluerobotics/BlueOS #1466: bug: VEHICLE id hardcoded to 1 (one)

mat2022, 2023-02-13:
### Bug description

Hey all,
I have to deploy a setup involving multiple vehicle. I am usingPi4/Navigator/BlueOS/custom ArduRover.
I think I found a “bug” in blueOS:
when you define sysid_thismav to something different from one (as I understand has to be the case in multi-vehicle setup), some strange things happen:
If you use BlueOS 1.1.0 beta 14, you get VEHICLE_TYPE_FETCH_FAIL and UPDATE_TIME_FAIL notification all the time.
If you use BlueOS 1.1.0 beta 11, you get VEHICLE_TYPE_FETCH_FAIL and FIRM_FETCH_FAIL notification all the time.

If you use the latest (older) stable release of BlueOS which is 1.0.1, you get no notification at all (great!).
BUT unfortunately, you still get the heartbeat lost (red broken heart) whereas there are plenty of heartbeat messages exchanged.

In all cases, as long as you define sysid_thismav to something not equal to one, you get the red broken heart, “heartbeat lost”.

I could find something in core/frontend/src/components/health/HealthTrayMenu.vue line 208:
what happens if system_id is not one ? like in a multi-vehicle setup ?
mounted() {
    mavlink2rest.startListening('HEARTBEAT').setCallback((message) => {
      if (message?.header.system_id !== 1 || message?.header.component_id !== 1) {   (this is line 208)
        return
      }

more specifically this system_id is used in different places:
command: 
grep -inr system_id
gives:
1 core/start-blueos-core:8:MAV_SYSTEM_ID=1
2 core/start-blueos-core:30:    'mavlink2rest',"mavlink2rest --connect=udpin:127.0.0.1:14000 --server 0.0.0.0:6040 --system-id $MAV_SYSTEM_ID --component-id $MAV_COMPONENT_ID_ONBOARD_COMPUTER4"
3 core/frontend/src/types/mavlink.ts:4:    system_id: number,
4 core/frontend/src/components/health/HealthTrayMenu.vue:155:      if (message?.header.system_id !== 1 || message?.header.component_id !== 1) {
5 core/frontend/src/libs/MAVLink2Rest/index.ts:101:        system_id: 255,
6 core/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py:17:        self.system_id = 7 int(os.environ.get("MAV_SYSTEM_ID", 1))
8 core/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py:23:    def set_system_id(self, system_id: int) -> None:
9 core/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py:24:        self.system_id = system_id
10 core/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py:74:            "header": {"system_id": self.system_id, "component_id": self.component_id, "sequence": self.sequence},

### Steps to reproduce

1 set SYSID_THISMAV to 2 (two) in QGroundControl
2 reboot vehicle as needed by this change.

3 Red Broken Heart meaning no more Heartbeat seen by interface.

4 Have a look to mavlink traffic proves plenty of HeartBeat messages exchanged.

### Primary pain point(s)

SYSID_THISMAV not equal to one is needed for multivehicle setup to function easily.
* point 1 and 2 may need investigation to ensure nothing bad will happen.
* problem of line 208 (point 4) could be temporarily fixed via:
if (message?.header.system_id > MAX_VEHICLE_ID || message?.header.component_id !== 1) {
whit this definition somewhere:
MAX_VEHICLE_ID = 240 (this guarantees some room for multi-vehicle (1...MAX_VEHICLE_ID) AND for multi GCS (MAX_VEHICLE_ID+1...255)
* I can give some of my time with enthusiasm, but I don't know anything about Docker, vue, javascript.
* I can do some testing with a proper guidance: typically if someone send me (a link to) a docker image, I can test it relatively easily

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-02-13:
Hi @mat2022, thanks for opening the issue.

It appears that the only place missing the usage of `MAV_SYSTEM_ID` is indeed on the frontend. We could use the [commander service](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/commander/main.py) to provide an API endpoint that gives all environment variables of the system, and with that, we could use such information to get the `SYSTEM_ID` of BlueOS and fix the if check on the frontend.

ES-Alexander, 2023-02-13:
Hi @mat2022, thanks for opening this :-)

I've added a bit of formatting to make your issue easier to parse, and added a link to our forum discussion for context.

Williangalvani, 2023-03-09:
extra thing to fix:
https://github.com/bluerobotics/BlueOS-docker/blob/master/core/libs/commonwealth/commonwealth/mavlink_comm/MavlinkComm.py#L44
we should make the default argument to vehicle self.system_id

closed 2023-09-23 by patrickelectric