bluerobotics/BlueOS #1489: Boot time improvements

Williangalvani, 2023-02-17:
We ran some tests yesterday:

rafaellehmkuhl, 2023-02-17:
Do we have a boot time profiling? RaspOS start time, Docker, Ardusub, services, etc.

Williangalvani, 2023-02-17:
> Do we have a boot time profiling? RaspOS start time, Docker, Ardusub, services, etc.

we can use `systemd-analyze` for overall pi boot time, `systemd-analyze critical-chain'  for the critical chain, and `systemd-analyze blame` for a list of times taken by individual services.
On top of that, core takes ~14 seconds to start after bootstrap is up

joaoantoniocardoso, 2023-02-17:
To read: https://silvertonconsulting.com/2016/05/06/faster-docker-initialization-through-slacker-snapshots-nfs-storage/

Williangalvani, 2023-08-04:
blueos_startup_update alone is taking roughly 10 seconds, which doesn't help.

patrickelectric, 2023-08-04:
> blueos_startup_update alone is taking roughly 10 seconds, which doesn't help.

We should add a version control system on that or something else to help us track when it's required to run.

patrickelectric, 2023-08-23:
from @joaoantoniocardoso `systemd-analyze plot > bootplot.svg`
[image]

patrickelectric, 2023-08-23:
Before #2020: boot time was 102s and moved to 90s (-12s)
Moving to podmand: boot time was 102s and moved to 71s (-31s)
Applying patches are taking 24s on log (python says 17s but it's wrong)

If we move to podman (-31s) and skip patch application (-24s) we can move BlueOS startup time to 35s.

joaoantoniocardoso, 2023-08-23:
> Before #2020: boot time was 102s and moved to 90s (-12s) Moving to podmand: boot time was 102s and moved to 71s (-31s) Applying patches are taking 24s on log (python says 17s but it's wrong)
> 
> If we move to podman (-31s) and skip patch application (-24s) we can move BlueOS startup time to 35s. Experiments done with @joaoantoniocardoso
> 
> > Tests done in a rasp pi 3

[image]

patrickelectric, 2023-08-23:
podman_service was responsible for starting BlueOS

joaoantoniocardoso, 2026-10-09:
In a Pi4, Bookworm, `1.4.7-beta.1`, sdcard 128GB:

| Event | Time |
|---|---|
| Reboot issued | 13.4 s (after the 5 s hold plus shutdown) |
| Bootloader banner | 13.9 s |
| `start4.elf` starts | 17.4 s |
| **Kernel starts** | **24.9 s** (firmware took 11.5 s) |
| Ethernet link up (kernel at 17.2 s) | about 42 s |
| Login prompt on the serial console | 41.7 s |
| **IP `192.168.2.2` assigned** (kernel at 40.5 s) | about 65 s |
| **Ping responds** | **65.6 s** |
| **SSH up** | **66.2 s** |
| **Web page up** | **66.3 s** |

AlejoPm03, 2026-10-31:
Comparison between 1.4.7 and master:
  
Both tests were run on a fresh SD card, after its first boot. Each test was repeated 5 times, and the median was used.

Except for the kernel and userspace metrics, all metrics were measured at the point where my PC could curl the endpoints or see and connect to the hotspot.

Setup:
- Pi 4
- SD 128GB
- Used CI-generated images [1.4.7](https://github.com/bluerobotics/BlueOS/releases/download/1.4.7/BlueOS-raspberry-linux-arm-v7-bullseye-pi4.zip)  [master](https://github.com/bluerobotics/BlueOS/actions/runs/40010242541/job/120896569144)

```
Metric                         1.4.7           master          Delta
-----------------------------------------------------------------------
Kernel                         2.0 (±0.0)      2.0 (±0.0)      +0.0
Userspace                      13.9 (±1.9)     13.9 (±1.2)     +0.0
nginx online                   66.3 (±3.6)     73.5 (±2.9)     +7.2
Frontend                       66.5 (±3.6)     73.5 (±2.9)     +7.0
Heartbeat (HTTP check)         66.7 (±3.6)     75.0 (±2.8)     +8.3
Hotspot visible                90.9 (±2.5)     97.9 (±3.4)     +6.9
Hotspot online (frontend)      97.1 (±2.6)     104.0 (±3.8)    +6.9
```