bluerobotics/BlueOS #1524: core: frontend: allow loading parameters from a log file

ES-Alexander, 2023-03-02:
### Current behaviour

The parameter editor allows loading parameters from a saved parameter file, but that's reliant on having a file backed up to restore/load from, which is not always ready in advance.

### Expected or desired behaviour

Parameters are generally recorded in autopilot logs, so it should be possible to add a "load from log" button to the parameter editor that allows selecting one of the available logs to extract and load the parameters from. This would be particularly useful for restoring to a previous 'known working' state with minimal advance preparation, especially in cases where it's known the date/time when a new feature/functionality started being tested.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-03-02:
Not sure if this is reliable since the log will contains the change of the parameters and not the parameters itself, is that correct ?

ES-Alexander, 2023-03-02:
I'm not certain - Willian mentioned this a couple of days ago.

That said, since the parameter editor (in BlueOS / QGC) requests all the parameters from the vehicle, if those messages are recorded then presumably it has a record of all of the values. IIRC Willian has already built this kind of functionality into the latest (master) version of Log Viewer, which I expect he wouldn't have done if the logs don't actually normally contain the parameters 🤷‍♂️

patrickelectric, 2023-03-02:
To document it better in the issue, It would be by processing mavlink data
to extract all parameters set / get message ? Since logviewer does that
already, it could extract the parameter file from there as a temporary
solution.

patrickelectric, 2023-03-02:
ping @Williangalvani

Williangalvani, 2023-03-02:
both logs, .bin and tlog, usually contains all parameters. .bin contains them by default.
tlogs require a gcs to request all parameters, but that is the case for all GCS I know so far.

And yeah, I don't consider this a high priority. the functionality is now in logviewer and should be useable as is.
I need to fix a couple of other bugs there and then I'll tag a new stable we can get into blueos.