bluerobotics/BlueOS #1525: core: frontend: provide a comparison of parameters when loading from file

ES-Alexander, 2023-03-02:
### Current behaviour

Loading parameters from a file currently saves them into the vehicle firmware, but does not provide an indication as to which (if any) parameters are being changed. This makes it unnecessarily inconvenient to understand which changes are being applied.

### Expected or desired behaviour

It would be helpful to provide some form of diff window when loading parameters, to show all the loaded parameters that differ from those already on the vehicle. Ideally this would be like QGC's comparison table, which allows checking/unchecking the parameters individually to determine which ones to include in the update (and which ones to leave as they are).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2023-03-02:
This could potentially share some functionality with #1245

patrickelectric, 2024-02-29:
I believe that is already done

closed 2024-02-29 by patrickelectric