bluerobotics/BlueOS #1530: We should have a better way to name individual cameras over putting the same name over multiple models

patrickelectric, 2023-03-02:
[image]

joaoantoniocardoso, 2023-03-02:
Would enumerating them be enough? "USB Camera 1" and "USB Camera 2" in this case.

ES-Alexander, 2023-03-03:
Daniel's actual comment (IIRC) was that it might make more sense to use the nickname (e.g. Aux Camera / Main Camera in this example) as the main titles (once a nickname has been assigned).

Using the device name there can be somewhat useful when trying to determine which camera is which while no stream has been created (especially since there isn't other information available), but once a stream is created it's a bit weird that the most prominent title for the card is the device type, rather than the name that the user has assigned for that device/stream.

There may be a bit of a semantic issue here if the intent is for one device to have multiple separate (named) stream cards going to different places in future, e.g. to handle different stream types (RTSP/UDP). If that is the plan then we should likely create an additional option to overwrite the device/stream name in the main title. If that's not the plan (and extra streams will be added within the existing stream card, like is currently the case for extra streams of the same type), then it seems to make sense to just replace the device/stream type title with the user's device/stream nickname once they've set up an output stream from that input.

ES-Alexander, 2023-03-03:
By the way @joaoantoniocardoso, what does the SDP file do when there are multiple streams configured for one input? Does it just include all of them and hope the receiving/playback application tries them until one works, or does it just only include the first one in the list or something?

joaoantoniocardoso, 2023-03-03:
> By the way @joaoantoniocardoso, what does the SDP file do when there are multiple streams configured for one input? Does it just include all of them and hope the receiving/playback application tries them until one works, or does it just only include the first one in the list or something?

The SDP is only using the first listed UDP address. Technically speaking, it's possible to add all the UDP addresses in a single SDP, I'm not sure if VLC has support for it, but we could also have one SDP per address.

joaoantoniocardoso, 2023-03-04:
> Using the device name there can be somewhat useful when trying to determine which camera is which while no stream has been created (especially since there isn't other information available), but [...]

I agree that this is not the easiest way to present the information. The current UI comes from the software architecture, definitely not a user-centric POV - there are sources and you can create (one) stream for those sources, which have a _(nick)name_, _encoding_, _resolution_, _framerate_, and might have more than one _endpoint_. For the developer, it's easier to be oriented by the _source_ instead of the _stream (nick)name_. 

However, being pragmatic, I don't yet see a simple way to change the main identifier of a card if we keep presenting the configured streams as child components of the video sources - this is, for me, the real problem. The plausible option to show the nickname instead of (or bigger than) the source if the card has a configured stream looks pretty confusing in my imagination, but still, it is a possibility.

Not long ago I was saying we should aggregate the cards by physical devices (device in the USB port 0), not by virtual device sources (/dev/video0, /dev/video1). But there's a better way to ease the usability - we could forget about sources and only show the configured streams on the main video page, if any. Then, we keep the sources reserved for the stream creation dialog plus an optional (deactivated by default) detailed view of the card of the configured streams.

joaoantoniocardoso, 2026-09-02:
Related: #2264