bluerobotics/BlueOS #1539: Allow changing network priority

patrickelectric, 2023-03-13:
When using a vehicle with multiple network interfaces: 4G, satellite, ethernet, wifi. We should allow the user to choose the priority between each one.

patrickelectric, 2023-06-21:
This appears to affect blue boat where internet access is not possible while the radio is on in some cases.

ES-Alexander, 2023-06-26:
It's be nice to also allow changing the priority [between wifi networks](https://www.raspberrypi.com/documentation/computers/configuration.html#adding-multiple-wireless-network-configurations). I'm not sure if that fits here or if it should be its own issue.

patrickelectric, 2023-06-26:
Yes, the idea is to show all network interfaces for the user to organize, exactly like macos.

ES-Alexander, 2023-06-26:
macOS standard network priority has “Wifi” as a single interface, which can be ordered relative to other interfaces like different ethernet ports.

Prioritisation of which wifi network to choose is handled separately, in an interface dedicated to just wifi selection, which is what I linked the equivalent of for raspberry pi. 

I’m not sure if you’re planning to “just” provide an interface for prioritisation between different network types (e.g. wifi/hotspot/usb-otg/Ethernet), or also allow prioritising between the known wifi networks.

patrickelectric, 2023-06-29:
> Prioritisation of which wifi network to choose is handled separately, in an interface dedicated to just wifi selection, which is what I linked the equivalent of for raspberry pi.

I'm not familiar to this interface, can you provide a screenshot ?
My idea is to show all interfaces.

ES-Alexander, 2023-06-29:
> I'm not familiar to this interface, can you provide a screenshot ?

It's apparently not possible anymore as of Ventura, but this is how it used to work:
https://www.howtogeek.com/532437/how-to-prioritize-wi-fi-networks-on-a-mac/

patrickelectric, 2023-08-04:
The network interface priority is changing after configuration

PR #1902 (2023-08-11): core: services: cable_guy: Move interface priority from ipr to dhcpcd
Fix #1539

Right now to have it working it's required to do a system reboot.
Doing dhcpcd service restart breaks our wpa_supplicant connection. #1958 was an attempt to fix that.
closed 2023-08-11 by patrickelectric