bluerobotics/BlueOS #1566: Extension-manager: Menu item not available after updating an extension

joaoantoniocardoso, 2023-03-29:
My BlueOS was with Cockpot v0.0.3 installed, and I could access it from the left menu. 

After update from v0.0.3 to v0.0.4, I can no longer access it from the left menu:
[image]

But after uninstalling it and installing again:
[image]

rafaellehmkuhl, 2023-03-29:
Duplicated of #1549 probably.

patrickelectric, 2023-03-29:
did you try to restart BlueOS ?

rafaellehmkuhl, 2023-04-11:
@Williangalvani @patrickelectric I can confirm this is happening for other extensions as well.

patrickelectric, 2023-04-11:
I have the same problem

Williangalvani, 2023-04-11:
same. it seems like the container is coming back with no ports
[image]

Williangalvani, 2023-04-11:
[this](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/kraken/kraken.py#L178) seems to be to blame

PR #1621 (2023-04-16): core: kraken: do not create user_permissions when updating an extension
fix #1566
closed 2023-04-16 by patrickelectric