bluerobotics/BlueOS #1600: Sometimes it reverts to factory after manual upload

joaoantoniocardoso, 2023-04-10:
Occasionally when I'm using the manual upload to test a PR, BlueOS reverts to the factory version.

patrickelectric, 2023-04-11:
From my research it appears that dealing with such files makes BlueOS unable to talk with bootstrap, resulting in a factory reset.

patrickelectric, 2023-04-27:
That's not only happening during the manual update process but also on normal use cases.
We are getting a number of reports that indicates that is something that is happening in the wild. 
We should:
  - Update version-chooser to provide logs
  - Allow user to update bootstrap
  - Allow user to configure bootstrap, like the factory time
  - Allow user to set "factory version"

joaoantoniocardoso, 2023-04-27:
We also need more swap. The following video was recorded during a manual upload:

https://user-images.githubusercontent.com/5920286/234951073-4d0315ce-9384-4840-a8c9-334f4ec3848a.mp4

joaoantoniocardoso, 2023-04-27:
We might need to configure the [Linux's OOM killer](https://docs.rackspace.com/support/how-to/linux-out-of-memory-killer/).

We can see here that it is acting:

[image]

patrickelectric, 2023-05-26:
Needs to be tested with Pi 4 2GB version and Pi 3 2GB version as well.

joaoantoniocardoso, 2023-05-26:
I can only confirm that it is happening on Raspberry Pi 3 w/ 1GB of ram.

joaoantoniocardoso, 2023-06-09:
Related to #1503

PR #3493 (2025-09-15): core: lift version chooser memory limit
fix #1600 

## Summary by Sourcery

Enhancements:
- Raise the memory allocation for the version chooser in start-blueos-core
closed 2025-09-15 by patrickelectric