bluerobotics/BlueOS #1614: allow manually uploading a docker image, for installing to offline / air-gapped systems (discussed here)

ES-Alexander, 2023-04-13:


ES-Alexander, 2024-07-01:
We'd need to add an API endpoint for loading an image (like we do [in version-chooser](https://github.com/bluerobotics/BlueOS/blob/master/core/services/versionchooser/utils/chooser.py#L79)), which then `import`s it into docker, before doing the normal custom extension registration process.

I propose we add a "manual upload" option to the custom registration interface for now, but once #2768 is complete there may be some other relevant interface options available too (e.g. if we want to support users installing official extension versions offline then we'd presumably want a streamlined way for them to be uploaded without needing custom registration, and possibly with [key-sign verification](https://github.com/bluerobotics/BlueOS/issues/2158) to confirm that they're actually legit images)

---

As an interim workaround it should be possible already to:
1. Upload the extension's docker image `.tar` ball to the BlueOS device, using either the File Browser or ssh
    - Probably makes sense to put it in the `/usr/blueos/extensions` folder (just `extensions` in the File Browser)
2. Use the BlueOS Terminal to [`docker import`](https://docs.docker.com/reference/cli/docker/image/import/#import-from-a-local-file) the image (may need to `red-pill` to be able to access the `docker` CLI)
3. [Tag](https://docs.docker.com/reference/cli/docker/image/tag/) the image with a tag that doesn't exist in the online registry
    - I'm unsure whether it will already have a tag when it's imported
4. Use the custom extension registration process to register that image + tag combo

rotu, 2024-10-04:
As an alternative to uploading a tarball, I prefer to use the docker engine on the rpi directly.

On your development machine do this once to create a context:
```shell
docker context create blueos
docker context update blueos --description "Docker Engine running on BlueOS"
docker context update blueos --docker "host=ssh://blueos.local"
docker --context=blueos buildx create --driver=docker-container --name=blueos-builder
docker --context=blueos buildx use blueos-builder --default
```

Switch your context (before you run docker commands) with:
```shell
docker context use blueos
```

Get all extension containers (i.e. name starts with `extension`) into a local `compose` file with 

```
docker run --rm -v /var/run/docker.sock:/var/run/docker.sock ghcr.io/red5d/docker-autocompose $(docker ps -aq --filter name=extension) > compose.yaml
```

Remove services from that file which you don't want to edit (e.g. `blueos-core`).

On the service you *want* to edit, replace the `image: ...` value with `build: /path/to/folder/containing/dockerfile/` 

Optionally, normalize your compose file with:

```
docker config --output compose.yaml
```

then edit and run with `docker compose up --build`

closed 2025-12-19 by patrickelectric