bluerobotics/BlueOS #1615: core: ardupilot-manager: support non-USB serial autopilot connections

ES-Alexander, 2023-04-14:
### Current behaviour

Flight controllers can currently only have a serial connection to the Raspberry Pi via a USB port. This unnecessarily prevents connecting an autopilot via a UART using the GPIO pins, which can be useful for space-saving, as well as general accessibility (more available interfaces -> higher chance of at least one of them being appropriate for a given application).

Issue first realised in [this forum post](https://discuss.bluerobotics.com/t/connect-rpi-to-ardu-using-gpio-uart/14055).

### Expected or desired behaviour

All serial ports should be considered as potential autopilot candidates - not just the ones identified as USB.

Initial attempt in #1580, although as per discussion there, that approach may only partially resolve the issue.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2023-04-21:
Also raised in [this forum post](https://discuss.bluerobotics.com/t/trying-to-connect-rpi-to-pixhawk-4-through-telem-1-2-port/14131).

ES-Alexander, 2023-04-24:
This is more important than we seem to have realised, because some firmwares apparently [correlate USB power with being on the bench](https://ardupilot.org/copter/docs/common-telemetry-port-setup.html#:~:text=the%20flight%20code%20assumes%20that%20if%20it%20is%20receiving%20power%20through%20this%20port%20that%20it%20is%20on%20the%20bench), and disable some failsafes as a result.

ES-Alexander, 2023-04-27:
Some comments [here](https://discuss.bluerobotics.com/t/trying-to-connect-rpi-to-pixhawk-4-through-telem-1-2-port/14131/9) on a recommended approach for adding a serial override option.

MrVoorakkara, 2024-08-02:
Are there any updates on this? Has there been any progress on a minimal override implementation?

patrickelectric, 2024-08-02:
Hi @MrVoorakkara which flight controller do you want to use and how ?

MrVoorakkara, 2024-08-02:
@patrickelectric I'm using a Holybro P6X running with a CM4 on their Pixhawk RPi CM4 Baseboard. Internally the autopilot is connected to the CM4 via Serial, Telem2 is connected to the Pi as below:

CM4 GPIO14 ↔ FMU TXD TELEM2

CM4 GPIO15 ↔ FMU RXD TELEM2

CM4 GPIO16 ↔ FMU CTS TELEM2

CM4 GPIO17 ↔ FMU RTS TELEM2

I would like BlueOS to leverage the available internal connection instead of having to use an external cable to connect the two, something which is space inefficient and unnecessarily takes up IO ports.

patrickelectric, 2024-08-02:
I see, thanks! We are going to look into it.

ES-Alexander, 2025-06-18:
[Raised in another forum post](https://discuss.bluerobotics.com/t/blueos-raspberry-pi-4b-5-and-cubepilot-orange-setup/20663/3).

ES-Alexander, 2026-02-06:
[Raised in another forum post](https://discuss.bluerobotics.com/t/help-to-enable-and-configure-rpi5-uart-pin-for-blueos-telem1-mavlink-communication/22413).