bluerobotics/BlueOS #1627: mavlink-camera-manager cpu usage is higher on latest master

Williangalvani, 2023-04-20:
Both when running an rtsp stream

Beta 16:
[image]

Master:
[image]

I've seen master at 140% cpu usage, but I'm unable to replicate.

patrickelectric, 2023-04-20:
maybe related with https://github.com/bluerobotics/mavlink-camera-manager/issues/215 ?
@joaoantoniocardoso

joaoantoniocardoso, 2023-04-20:
> maybe related with [bluerobotics/mavlink-camera-manager#215](https://github.com/bluerobotics/mavlink-camera-manager/issues/215) ? @joaoantoniocardoso

unrelated to that, possibly related to (a) thumbnails or (b) proxies.

@Williangalvani was you with BlueOS' video page opened?

Williangalvani, 2023-04-24:
> So that's probably (b) proxies. I'd need to confirm with tests, but that's really bad news

what do you mean by proxies? inside mcm?

joaoantoniocardoso, 2023-04-24:
From your graphs I see two interesting things:
- higher jitter in delivered frame rate, with huge dips, which are freezings
- doubled the CPU usage

Probably the higher CPU is causing the freezings, which doesn't seem to be related to the thumbnail system.

---

Just started an investigation here... Looking for causes for that higher CPU usage. Beta16 uses MCM t3.10.1 ([diff](https://github.com/bluerobotics/mavlink-camera-manager/compare/t3.10.1...t3.11.3))

From that diff, I found only three possible related changes:
- [using proxysrc/proxyxsink pair to split each sink into a different pipeline (UDP and ImageSInk only) 
](https://github.com/bluerobotics/mavlink-camera-manager/compare/t3.10.1...t3.11.3#diff-2f1f6625bb4a423e0d90240558d364879a8032f93f342a9b65b2321432752869R15-R16)
- [no more fixating the profile](https://github.com/bluerobotics/mavlink-camera-manager/compare/t3.10.1...t3.11.3#diff-aba234205cda9893ce54b18bd8e05da0a3c73b9bc0afc514f3ab7a4ad0c61025R57)
- [pipeline runner now runs on in tokio green thread instead of in an OS thread](https://github.com/bluerobotics/mavlink-camera-manager/compare/t3.10.1...t3.11.3#diff-c9de2f4f0f6a1630959cbe0944fc9a1ae9e1c20d6d01a32faf0c28b5f1a2cc68R139)

Obviously, splitting means more pipelines, more threads, and more overhead, but it's probably something heavier than that happening on the GStreamer side. I'd try to replicate and investigate it further.

joaoantoniocardoso, 2023-04-24:
[image]

This queue4 is the one that connects the Tee from the v4l source pipeline with the webrtcbin

joaoantoniocardoso, 2023-04-24:
We have a clue here, after running for some time:

[image]

joaoantoniocardoso, 2023-04-25:
:facepalm: `current-level-time=992125648` means it's holding almost 1 second of data. To make it stateless I should also set `max-size-time=0`. I'd already noticed that when working on the RTSP to WebRTC, so a Patch is on the way...

joaoantoniocardoso, 2023-04-26:
Turns out that the problem was something completely different: an integration in a logging library, anyways, it's fixed now :)

PR #1651 (2023-04-27): core: tools: mavlink-camera-manager: Update to t3.11.4.
Closes #1627

[Release notes](https://github.com/bluerobotics/mavlink-camera-manager/releases/tag/t3.11.4)
closed 2023-04-27 by patrickelectric