bluerobotics/BlueOS #1640: File system: explicitly support extensions

ES-Alexander, 2023-04-22:
We should have a part of the file system dedicated to persistent data for extensions, such as configuration, logs, icons/logos, readme, and relevant user data.

Having a folder per extension would allow consistent binding rules, monitoring (to ensure data usage remains within user-specified limits), and clearing (e.g. when uninstalling, or as a refresh), and a fixed location (and filename structure?) for logs would make them easier to access and manage.

I recommend adding a shortcut (like the existing ones in the file browser), with `extensions` as the main folder, then `docker-image-name` for the individual extensions, which should include a `logs` folder for any logs, a readme file, an icon image file, a `logos` folder for company logos and the like (in case the user wants to include them in the upcoming logos section of the BlueOS interface), and anything else the extension wants to store.

Relevant to:
- #1611 
- #1441 
- #1243
- #1231 
- #1180 (logos and readmes)

Williangalvani, 2023-04-24:
That is a very good idea. We could also mount the storage folder automatically, which would make the json simpler for extensions using it for trivial things

ES-Alexander, 2024-01-26:
> ... We could also mount the storage folder automatically, which would make the json simpler for extensions using it for trivial things

This does seem convenient, but we'll need to think about a reasonable binding location within the extension - e.g. binding to root seems problematic, in case there are things the developer doesn't actually want accessible from outside the extension.

My QuickStart Python extension currently [binds to `/app`](https://github.com/ES-Alexander/QuickStart-Python-Extension/blob/main/Dockerfile#L18), but I'm not sure if there's some better / more generic idea that would be preferable instead.