bluerobotics/BlueOS #1642: [Kraken] -> NS_BINDING_ABORTED -> No Extensions available. Make sure the vehicle has internet access and try again.

joaoantoniocardoso, 2023-04-24:
[image]

joaoantoniocardoso, 2023-04-24:
log:
```
2023-04-24 14:43:05.041 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56264 - "GET / HTTP/1.1" 200
2023-04-24 14:43:05.052 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56266 - "GET /register_service HTTP/1.1" 404
2023-04-24 14:43:05.061 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56268 - "GET /docs HTTP/1.1" 200
2023-04-24 14:43:05.071 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56270 - "GET /docs.json HTTP/1.1" 404
2023-04-24 14:43:05.081 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56272 - "GET /openapi.json HTTP/1.1" 200
2023-04-24 14:43:05.092 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56274 - "GET /v1.0/docs HTTP/1.1" 200
2023-04-24 14:43:05.102 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56276 - "GET /v1.0/openapi.json HTTP/1.1" 200
2023-04-24 14:43:05.112 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56278 - "GET / HTTP/1.1" 200
2023-04-24 14:43:05.121 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:56280 - "GET /swagger.json HTTP/1.1" 404
2023-04-24 14:43:17.810 | ERROR    | commonwealth.utils.apis:custom_route_handler:36 - Unhandled service exception.
2023-04-24 14:43:17.811 | ERROR    | commonwealth.utils.apis:custom_route_handler:37 - 
Traceback (most recent call last):

  File "/home/pi/services/kraken/main.py", line 179, in <module>
    loop.run_until_complete(server.serve())
    │    │                  │      └ <function Server.serve at 0xb5c7f610>
    │    │                  └ <uvicorn.server.Server object at 0xb568e178>
    │    └ <function BaseEventLoop.run_until_complete at 0xb63d5bf8>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>

  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 634, in run_until_complete
    self.run_forever()
    │    └ <function BaseEventLoop.run_forever at 0xb63d5bb0>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 601, in run_forever
    self._run_once()
    │    └ <function BaseEventLoop._run_once at 0xb63d7970>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1905, in _run_once
    handle._run()
    │      └ <function Handle._run at 0xb641f658>
    └ <Handle <TaskStepMethWrapper object at 0xb55835b0>()>
  File "/usr/local/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
    │    │            │    │           │    └ <member '_args' of 'Handle' objects>
    │    │            │    │           └ <Handle <TaskStepMethWrapper object at 0xb55835b0>()>
    │    │            │    └ <member '_callback' of 'Handle' objects>
    │    │            └ <Handle <TaskStepMethWrapper object at 0xb55835b0>()>
    │    └ <member '_context' of 'Handle' objects>
    └ <Handle <TaskStepMethWrapper object at 0xb55835b0>()>
  File "/usr/local/lib/python3.9/site-packages/uvicorn-0.13.4-py3.9.egg/uvicorn/protocols/http/h11_impl.py", line 396, in run_asgi
    result = await app(self.scope, self.receive, self.send)
                   │   │    │      │    │        │    └ <function RequestResponseCycle.send at 0xb5651100>
                   │   │    │      │    │        └ <uvicorn.protocols.http.h11_impl.RequestResponseCycle object at 0xb55aa220>
                   │   │    │      │    └ <function RequestResponseCycle.receive at 0xb5651148>
                   │   │    │      └ <uvicorn.protocols.http.h11_impl.RequestResponseCycle object at 0xb55aa220>
                   │   │    └ {'type': 'http', 'asgi': {'version': '3.0', 'spec_version': '2.1'}, 'http_version': '1.0', 'server': ('127.0.0.1', 9134), 'cl...
                   │   └ <uvicorn.protocols.http.h11_impl.RequestResponseCycle object at 0xb55aa220>
                   └ <uvicorn.middleware.proxy_headers.ProxyHeadersMiddleware object at 0xb568e268>
  File "/usr/local/lib/python3.9/site-packages/uvicorn-0.13.4-py3.9.egg/uvicorn/middleware/proxy_headers.py", line 45, in __call__
    return await self.app(scope, receive, send)
                 │    │   │      │        └ <bound method RequestResponseCycle.send of <uvicorn.protocols.http.h11_impl.RequestResponseCycle object at 0xb55aa220>>
                 │    │   │      └ <bound method RequestResponseCycle.receive of <uvicorn.protocols.http.h11_impl.RequestResponseCycle object at 0xb55aa220>>
                 │    │   └ {'type': 'http', 'asgi': {'version': '3.0', 'spec_version': '2.1'}, 'http_version': '1.0', 'server': ('127.0.0.1', 9134), 'cl...
                 │    └ <uvicorn.middleware.message_logger.MessageLoggerMiddleware object at 0xb568e1c0>
                 └ <uvicorn.middleware.proxy_headers.ProxyHeadersMiddleware object at 0xb568e268>
  File "/usr/local/lib/python3.9/site-packages/uvicorn-0.13.4-py3.9.egg/uvicorn/middleware/message_logger.py", line 61, in __call__
    await self.app(scope, inner_receive, inner_send)
          │    │   │      │              └ <function MessageLoggerMiddleware.__call__.<locals>.inner_send at 0xb56c3190>
          │    │   │      └ <function MessageLoggerMiddleware.__call__.<locals>.inner_receive at 0xb56c3b68>
          │    │   └ {'type': 'http', 'asgi': {'version': '3.0', 'spec_version': '2.1'}, 'http_version': '1.0', 'server': ('127.0.0.1', 9134), 'cl...
[... cut ...]
2023-04-24 14:47:25.283 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:60514 - "GET / HTTP/1.1" 200
2023-04-24 14:47:25.292 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:60516 - "GET /swagger.json HTTP/1.1" 404
2023-04-24 14:47:51.543 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:60872 - "GET /v1.0/stats HTTP/1.0" 200
2023-04-24 14:47:51.573 | INFO     | uvicorn.protocols.http.h11_impl:send:461 - 127.0.0.1:60932 - "GET /v1.0/list_containers HTTP/1.0" 200

```

joaoantoniocardoso, 2023-04-24:
After manually resetting the service:

[image]

Williangalvani, 2023-04-24:
The timeout makes me believe this was a networking issue. is it easily replicated?

joaoantoniocardoso, 2023-04-24:
> The timeout makes me believe this was a networking issue. is it easily replicated?

Not _easy_, BlueOS restart and container restart don't cause it, but it has been annoyingly happening just when I need to manage an extension -- pretty sure the chances are higher if I'm in a rush :laughing:

joaoantoniocardoso, 2023-08-04:
I haven't seen this since then, so it's probably already fixed.

closed 2023-08-04 by joaoantoniocardoso