bluerobotics/BlueOS #1660: [Guide] How to upgrade from older BlueOS

joaoantoniocardoso, 2023-05-01:
## Problem

Older BlueOS versions shipping older blueos-bootstrap (like [beta15](https://hub.docker.com/layers/bluerobotics/blueos-bootstrap/1.1.0-beta.15/images/sha256-3611577a7421d1413e911cd3f54d23fbc34b6cfdce17e6f654a89915c8297cb2?context=explore)) can have problems when trying to upgrade to more recent BlueoS versions, say beta20 or the current master.

The video below shows an attempt in a Raspberry Pi 3, with a beta15 factory (with beta15 bootstrap):

https://user-images.githubusercontent.com/5920286/235496239-14fd4868-f0bd-4b91-ad92-a3955fe710e6.mp4

## Solution

To solve this issue, you need to upgrade blueos-bootstrap to [beta20](https://hub.docker.com/layers/bluerobotics/blueos-bootstrap/1.1.0-beta.20/images/sha256-76de1b29fc2b779b8e5b73a6b3ecc5814ba7d7c0acda090644c77fe714054d1f?context=explore) or a higher version, and then restart your Raspberry Pi. You can achieve this by following these steps:

 1. Upgrade blueos-bootstrap to beta20 or a higher version.
 2. Restart your Raspberry Pi.

To do so, the following command sequence can be run in the Raspberry Pi terminal (outside BlueOS -- when you `red-pill`):

``` Bash
export BLUEOS_BOOTSTRAP=bluerobotics/blueos-bootstrap:1.1.0-beta.20 ;
docker stop blueos-bootstrap &&
docker rm blueos-bootstrap &&
docker pull $BLUEOS_BOOTSTRAP &&
docker create \
    -t \
    --restart unless-stopped \
    --name blueos-bootstrap \
    --net=host \
    -v $HOME/.config/blueos/bootstrap:/root/.config/bootstrap \
    -v /var/run/docker.sock:/var/run/docker.sock \
    -e BLUEOS_CONFIG_PATH=$HOME/.config/blueos \
    $BLUEOS_BOOTSTRAP ;
unset BLUEOS_BOOTSTRAP ;
sudo reboot
```

To simplify it, the following command was tailored using the commander to be executed from the top-side computer, considering `blueos.local` as the BlueOS address:
``` Bash
curl -X 'POST' \
  'http://blueos.local:9100/v1.0/command/host?command=export%20BLUEOS_BOOTSTRAP%3Dbluerobotics%2Fblueos-bootstrap%3A1.1.0-beta.20%20%3B%20docker%20stop%20blueos-bootstrap%20%26%26%20docker%20rm%20blueos-bootstrap%20%26%26%20docker%20pull%20%24BLUEOS_BOOTSTRAP%20%26%26%20docker%20create%20%5C%20%20%20%20%20-t%20%5C%20%20%20%20%20--restart%20unless-stopped%20%5C%20%20%20%20%20--name%20blueos-bootstrap%20%5C%20%20%20%20%20--net%3Dhost%20%5C%20%20%20%20%20-v%20%24HOME%2F.config%2Fblueos%2Fbootstrap%3A%2Froot%2F.config%2Fbootstrap%20%5C%20%20%20%20%20-v%20%2Fvar%2Frun%2Fdocker.sock%3A%2Fvar%2Frun%2Fdocker.sock%20%5C%20%20%20%20%20-e%20BLUEOS_CONFIG_PATH%3D%24HOME%2F.config%2Fblueos%20%5C%20%20%20%20%20%24BLUEOS_BOOTSTRAP%20%3B%20unset%20BLUEOS_BOOTSTRAP%20%3B%20sudo%20reboot&i_know_what_i_am_doing=true' \
  -H 'accept: application/json' \
  -d ''
```

Be aware that it will take a while because the image will be downloaded (approx. 36MB) from the Dockerhub servers to the Pi.

After running the commands, it succeeds:

https://user-images.githubusercontent.com/5920286/235496537-922524a7-2c36-42b3-8af5-10f8024f477d.mp4

joaoantoniocardoso, 2023-05-05:
It's not always a success:

``` bash
pi@blueos:~ $ docker pull bluerobotics/blueos-core:1.1.0-beta.20
1.1.0-beta.20: Pulling from bluerobotics/blueos-core
b4770fcf2ef7: Already exists 
66af9b3c4eb3: Already exists 
c3df7581cc70: Already exists 
46b0bc6ef984: Already exists 
c967989bced8: Already exists 
3b5d66328499: Extracting [==================================================>]  1.161kB/1.161kB
994dc793e8c7: Download complete 
073b491e2ef0: Download complete 
0b9c1ec5f936: Download complete 
9ce7e5e3a08a: Download complete 
84eb2bf9b357: Download complete 
84c0f950f365: Download complete 
4ca1fad7c5dd: Download complete 
f4ce1277b4b8: Download complete 
91c7d6067f1c: Download complete 
89f4ef418459: Download complete 
a1ba11ca18b8: Download complete 
ee7762c77d20: Download complete 
d6c387134483: Download complete 
b3319e86b204: Download complete 
10e3537725db: Download complete 
e2f0dfd80a21: Download complete 
8dbaf581470d: Download complete 
fe1aaa9f8dce: Download complete 
c1b58e43ce80: Download complete 
9e0bf7be7e71: Download complete 
cee2287a6621: Download complete 
4f4fb700ef54: Download complete 
4ba7abb72482: Download complete 
5bfdc2426874: Download complete 
372ef151c13b: Download complete 
6f59f50c62fd: Download complete 
377249ca3c61: Download complete 
99e8684a5d3c: Download complete 
b1b124f11fd2: Download complete 
failed to register layer: error creating overlay mount to /var/lib/docker/overlay2/efa827d801e3dea9ba28383bfb48768c80e8dbeab5bb95abac7b5fb4e067bd99/merged: too many levels of symbolic links
pi@blueos:~ $ 
```

joaoantoniocardoso, 2023-05-27:
@patrickelectric what does mean this guide as part of the milestone? :eyes:

patrickelectric, 2023-05-27:
We should check and make sure that 1.0.1 to 1.1 works without problems