bluerobotics/BlueOS #1662: ParameterEditor: Save button does not get enabled after changing item in list

patrickelectric, 2023-05-02:
a parameter such as ARMING_CHECK fails to enable the save button once edited

Williangalvani, 2023-05-04:
related to #1562

closed 2023-05-07 by patrickelectric

clydemcqueen, 2023-08-25:
I'm seeing this behavior again on beta 28. I can't change LOG_DISARMED from Disabled to Enabled.

I also tried to change it via "custom" and it did not work.

This seems to affect all parameters.

[image]

BlueOS Version: [tags/1.1.0-beta.28-0-g2df6ae03](https://github.com/bluerobotics/blueos-docker/releases/tag/1.1.0-beta.28)
Bootstrap Version: 1.0.1
Build: 8/12/2023, 8:51:38 PM
By [Blue Robotics](https://bluerobotics.com/)

Board name:
Pixhawk1
Manufacturer:
ArduPilot
Mavlink platform:
Pixhawk1
Firmware version:
4.1.0 (STABLE)
Vehicle type:
Submarine
Path:
/dev/ttyACM0