bluerobotics/BlueOS #1663: Current (fixed) SITL command line parameters do not work

rafaellehmkuhl, 2023-05-03:
If you try to use a rover firmware, for example, the vehicle will not move correctly because the SITL is using VECTORED as a fixed default frame.

To fix that we can allow the SITL frame to be configured.

Another cool idea from @Williangalvani is to just allow the user to send a string with the parameters that he wants and just override everything with it, so the user gain full control.

ES-Alexander, 2026-01-09:
As a workaround:

> Go to the Available Services page, and click the "v2.0 docs" link for Autopilot Manager (in the far right column). On the page that loads, under `/sitl_frame` change the frame type to motorboat and Execute:
> [image]

patrickelectric, 2026-09-11:
Fixed in #3950, the frame is configurable in the SITL panel.

closed 2026-09-11 by patrickelectric