bluerobotics/BlueOS #1674: Vehicle (on pwm outputs tab) very hard to see in dark mode

rafaellehmkuhl, 2023-05-05:
[image]

A lighter background would make it easier to see.

patrickelectric, 2023-05-07:
Looks fine here:
[image]
[image]

Are you using an extension or something ?
Chrome on mac without dark/light theme extension manager.

rafaellehmkuhl, 2023-05-08:
It happens here in both BlueOS dark and light modes.
I'm using Chrome on a Mac with Apple chip and my browser is in light mode.

[image]

closed 2026-09-03 by joaoantoniocardoso