bluerobotics/BlueOS #1695: bug: No Extensions available.

phamquandung, 2023-05-23:
### Bug description

No Extensions available. Make sure the vehicle has internet access and try again.

### Steps to reproduce

1. Access: http://192.168.2.2/tools/extensions-manager
2. The error pop-up: No Extensions available. Make sure the vehicle has internet access and try again.

I check the internet, all work file. I tried on BlueOS version 1.1.0-beta-21 and 18. And also for Remote Versions: error fetching online tags: Cannot connect to host auth.docker.io:443 ssl:default [Temporary failure in name resolution]

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2023-05-23:
Hi @phamquandung,

Thanks for reporting the issue, is this a consistent problem ? Is it working fine now ?

phamquandung, 2023-05-23:
> Hi @phamquandung,
> 
> Thanks for reporting the issue, is this a consistent problem ? Is it working fine now ?
Hello @patrickelectric . It's still there. Please check
[image]

patrickelectric, 2023-05-23:
Please connect your vehicle to wifi, check the wifi icon.
https://docs.bluerobotics.com/ardusub-zola/software/onboard/BlueOS-1.1/getting-started/#connect-wifi

patrickelectric, 2026-08-27:
It appears that Docker can run only with ipv6 in some cases.

```
getent hosts auth.docker.io hub.docker.com

2a06:98c1:3104::6812:2bb2 auth.docker.io.cdn.cloudflare.net auth.docker.io
2606:4700:4403::ac40:904e auth.docker.io.cdn.cloudflare.net auth.docker.io
```

This appears to be one of the problems that results in:

[image]

patrickelectric, 2026-08-27:
Fixed the fetch but not the download, working on it:

```
The operation failed: error while pulling image: DockerError(500, 'Get "https://registry-1.docker.io/v2/": dial tcp [2600:1f18:2148:bc01:e939:a0be:6929:ed84]:443: connect: network is unreachable')
```