bluerobotics/BlueOS #1704: frontend: links to extension are broken

Williangalvani, 2023-05-24:
[image]

Williangalvani, 2023-05-25:
try cockpit:
[Screencast from 2023-05-25 13-53-12.webm](https://github.com/bluerobotics/BlueOS-docker/assets/4013804/3e070a0c-3c0b-42cd-9f59-8dce644f9458)

joaoantoniocardoso, 2023-05-26:
still broken, but in a different way

[image]

Williangalvani, 2023-05-26:
@joaoantoniocardoso can you double check your master is docker 4dab309e? git BlueOS Version: [heads/master-0-gd5287b99](https://github.com/bluerobotics/blueos-docker/tree/d5287b99)

joaoantoniocardoso, 2023-05-26:
My bad, this was indeed solved by that PR, I was testing the yesterday's image.

PR #1716 (2023-05-26): core: frontend: App: Fix access to extensions that open in new pages
Fix #1704
closed 2023-05-26 by joaoantoniocardoso